robocasa/robocasa #12: Behavior cloning training scripts?

cheryyunl, 2024-06-01:
Hi there,

We plan to use Robocasa as an experimental domain for our paper. However, we're concerned that our current behavior cloning (BC) training script might not align perfectly with the Robocasa benchmark requirements.

Would you consider releasing your BC training code? Having it as a reference would be immensely helpful for us to ensure compatibility and achieve the best possible results with Robocasa.

Thanks in advance for your time and consideration!

snasiriany, 2024-06-03:
Please see our policy learning repo! https://github.com/ARISE-Initiative/robomimic/tree/robocasa

Note, this repo is experimental, and we are still in the process of uploading all of our MimicGen datasets.

closed 2024-06-03 by snasiriany

luccachiang, 2024-06-11:
Hi @snasiriany , would you please check this issue? It seems that there is something wrong with the policy learning repo. Thanks! https://github.com/ARISE-Initiative/robomimic/issues/159

snasiriany, 2024-06-11:
Yes! Following up in that thread