robocasa/robocasa #14: Question about running RoboCasa in a headless setting

MARMOTatZJU, 2024-06-07:
Thank you for this gorgeous simulator for the community! I would like to know if it is possible run RoboCasa on a headless server (i.e. forward window from server to local) and how to achieve it.

Thanks in advance!

MARMOTatZJU, 2024-06-14:
> Hi @MARMOTatZJU are you looking to use a GUI window through a remote connection? Or are you simply looking to render image "offscreen" without a GUI window?

I would appreciate it if any clue about both cases could be provided. Thanks~

snasiriany, 2024-06-16:
Certainly, here's an example for offscreen rendering:

```
import robocasa
import robosuite
from robosuite import load_controller_config

env = robosuite.make(
    env_name="PnPCounterToCab",
    robots="PandaMobile",
    controller_configs=load_controller_config(default_controller="OSC_POSE"),
    # add additional env params here as needed
    
    # rendering settings
    has_renderer=False,
    has_offscreen_renderer=True,
    use_camera_obs=True,
    camera_depths=False, # set to True to also render depth
    camera_names=["robot0_agentview_left", "robot0_agentview_right", "robot0_eye_in_hand"],
)

obs = env.reset()

### access images, eg. obs["robot0_agentview_left_image"] ###
```

You can also render these images through a remote connection via [X11](https://hrs4real.medium.com/mastering-real-time-video-streaming-on-linux-a-guide-to-python-and-ssh-x11-forwarding-ffce1fb0130c).

closed 2024-06-16 by snasiriany