robocasa/robocasa #16: Issue about ground truth point cloud

fujiawei0724, 2024-06-13:
Thanks the authors for the great work. I have a tiny question, in the scenarios of the simulator, can I get the ground truth point cloud with instance labels of every object? Looking forward to your reply

snasiriany, 2024-06-17:
Hi @fujiawei0724, we do support depth rendering, you can enable it by setting `camera_depths=True` [here](https://github.com/robocasa/robocasa/blob/1370b9e0f747d84fb21ed29bacefb1654865301b/robocasa/environments/kitchen/kitchen.py#L65).

Here's an example script to render depth images, written by @Abhiram824:

```
import robocasa
import robosuite
from robosuite import load_controller_config
import numpy as np
import matplotlib.pyplot as plt

env = robosuite.make(
    env_name="PnPCounterToCab",
    robots="PandaMobile",
    controller_configs=load_controller_config(default_controller="OSC_POSE"),
    
    has_renderer=False,
    has_offscreen_renderer=True,
    use_camera_obs=True,
    camera_depths=True,
    camera_names="robot0_agentview_center",
)
obs = env.reset()
image = obs["robot0_agentview_center_image"][::-1]
depth_map = obs["robot0_agentview_center_depth"][::-1]
# print(depth.shape)

depth_map = np.squeeze(depth_map)

fig, ax = plt.subplots(1, 2, figsize=(12, 6))

ax[0].imshow(image)
ax[0].set_title('Image')
ax[0].axis('off')  

depth_map_display = ax[1].imshow(depth_map, cmap='viridis')
ax[1].set_title('Depth Map')
ax[1].axis('off')  

fig.colorbar(depth_map_display, ax=ax[1], fraction=0.046, pad=0.04)

plt.savefig('depth_map.png')
```

Result: 
[image]

You can use external libraries to get pointclouds from the rgb and depth images, here is one possible solution: https://stackoverflow.com/a/59590281

At the moment we do not have support for segmentations, but will add it to our bucket list for future releases.

closed 2024-06-17 by snasiriany

karthikm-0, 2024-08-26:
Are there any updates on segmentation? If not, do you have any advice for implementing this? I could prototype and pull it if so.

snasiriany, 2024-08-26:
@abhihjoshi is working on this at the moment. See here: https://github.com/robocasa/robocasa/issues/43

karthikm-0, 2024-08-26:
Thanks for pointing me to it!

LeiHanyue, 2025-03-05:
What are the units of depth in this depth map? Why does it feel like the depth values in this image are unreasonable?

EricYang3721, 2025-04-08:
This might be the solution for above question https://github.com/ARISE-Initiative/robosuite/issues/516