robocasa/robocasa #18: How to load different robot models in tele-operation script?

vantony1, 2024-06-14:
Hello,

Thank you for answering my previous question, I appreciate the help! 

I have continued to work with the robocasa simulation and am trying to see to what extent we can use it for my current research project.

I was wondering how I can load the different embodiments the simulation supports in the demo teleoperation script?

https://github.com/robocasa/robocasa/blob/2918ef92925d5ac8a7baf57f9b0186c2049f48fb/robocasa/demos/demo_teleop.py#L94

It seems like I should manipulate the value here ^ however it is unclear to me what values are accepted. Robosuite supports a range of robots (https://robosuite.ai/docs/modules/robots.html#manipulators) however it seems like these don't correspond directly with the mobile manipulators that your paper/website referred to.

Could you please advise me on this? 

Thanks!

Xianqi-Zhang, 2024-06-19:
Same question.

Even names mentioned elsewhere in the code cannot work, such 'Panda' and 'Sawyer'.
Robots mentioned in robosuite robocasa_v0.1 branch also do not work.
It's really hard to use !

https://github.com/robocasa/robocasa/blob/1370b9e0f747d84fb21ed29bacefb1654865301b/robocasa/scripts/tune_joints.py#L254
https://github.com/robocasa/robocasa/blob/1370b9e0f747d84fb21ed29bacefb1654865301b/robocasa/scripts/tune_camera.py#L122
```python
 # * https://github.com/ARISE-Initiative/robosuite/blob/29ee7efb49476d5bcc5bb43cf75105903eadb1a7/robosuite/robots/__init__.py#L11-L35
ROBOT_CLASS_MAPPING = {
    "Baxter": FixedBaseRobot,
    "IIWA": FixedBaseRobot,
    "Jaco": FixedBaseRobot,
    "Kinova3": FixedBaseRobot,
    "Panda": FixedBaseRobot,
    "Sawyer": FixedBaseRobot,
    "UR5e": FixedBaseRobot,
    "VX300S": FixedBaseRobot,
    "Z1": FixedBaseRobot,
    "PandaMobile": WheeledRobot,
    "GoogleRobot": WheeledRobot,
    "VX300SMobile": WheeledRobot,
    "Tiago": WheeledRobot,
    "B1Z1": LeggedRobot,
    # "B1Z1Floating": LeggedRobot,
    "GR1": LeggedRobot,
    "GR1FixedLowerBody": LeggedRobot,
    "GR1ArmsOnly": LeggedRobot,
    "GR1FloatingBody": LeggedRobot,
    "H1": LeggedRobot,
    "H1FixedLowerBody": LeggedRobot,
    "H1FloatingBody": LeggedRobot,
    "H1ArmsOnly": LeggedRobot,
}
```

snasiriany, 2024-06-20:
Hi @vantony1 @Xianqi-Zhang, our public repo at the moment has limited support for cross embodiment. While we do have the robot models for all of these robots, we still need to update some logic, controller parameters, and also provide more in-depth documentation (which we are in the process of making right now). This is one of our biggest priorities and we are planning to complete these items in the coming weeks.

I'll share an update here when this is complete.

Xianqi-Zhang, 2024-06-22:
> Hi @vantony1 @Xianqi-Zhang, our public repo at the moment has limited support for cross embodiment. While we do have the robot models for all of these robots, we still need to update some logic, [...]

Thank you very much for your reply. 
Looking forward to using the same or even more powerful env as mentioned in the paper, especially the part about humanoid robots. 
In addition, if possible, please add more baseline codes and results (e.g., IL and RL) to facilitate further research. 
Thank you very much and will continue to pay attention.

zhangjb416, 2024-09-02:
> Hi @vantony1 @Xianqi-Zhang, our public repo at the moment has limited support for cross embodiment. While we do have the robot models for all of these robots, we still need to update some logic, [...]

Thanks for your great work and effort in releasing the dataset and benchmark! I am also looking forward to explore the powerful envirionment and the cross-embodiment deployment. May I inquire if you have any plans to share the updates in the near future?

Thanks a lot!

snasiriany, 2024-09-03:
We are still working on this. We are rolling out updates regularly to expand the set of robot models and to make them more modular. You can view the list of robots here: https://github.com/ARISE-Initiative/robosuite/tree/robocasa_v0.1/robosuite/models/assets/robots

We are working on adding official support for some of these robots in RoboCasa. We will release some updates later this week on this front.

tjvitchutripop, 2024-10-09:
Any updates on this @snasiriany?  Would love to be able to use the quadruped + arm setup seen in the paper.

snasiriany, 2024-10-10:
It's almost ready. We have been working hard at a major refactoring of RoboSuite that will also benefit all RoboCasa users. This is going to be released around early November.

BoyuLi-XH, 2024-12-05:
I am looking forward to the progress of the humanoid robot simulation environment. Will there be any updates soon? @snasiriany

snasiriany, 2024-12-05:
Hi @BoyuLi-XH the humanoid robot is ready. Please run
```
python ~/code/robocasa-dev/robocasa/scripts/collect_demos.py --debug --env <taskname> --robots GR1FixedLowerBody
```

BoyuLi-XH, 2024-12-09:
> Hi @BoyuLi-XH the humanoid robot is ready. Please run
> 
> ```
> python ~/code/robocasa-dev/robocasa/scripts/collect_demos.py --debug --env <taskname> --robots GR1FixedLowerBody
> ```

Thanks a lot.

BoyuLi-XH, 2024-12-11:
> Hi @BoyuLi-XH the humanoid robot is ready. Please run
> 
> ```
> python ~/code/robocasa-dev/robocasa/scripts/collect_demos.py --debug --env <taskname> --robots GR1FixedLowerBody
> ```

Thanks for your kind response. But I meet some mistakes run this code. I run 
 ```
 python ./robocasa/scripts/collect_demos.py --debug --env PrepareCoffee --robots GR1FixedLowerBody
 ```
The result is 
 ```
WARNING: robosuite environments not imported...
Got error: No module named 'robosuite.environments.manipulation.single_arm_env'
WARNING: robosuite task zoo environments not imported, possibly because robosuite_task_zoo is not installed...
Got error: No module named 'robosuite.environments.manipulation.single_arm_env'
[robosuite INFO] Loading controller configuration from: /home/ps/Desktop/lby_workspace/robosuite/robosuite/controllers/config/robots/default_gr1_fixed_lower_body.json (composite_controller_factory.py:121)
Opening SpaceMouse device
[robosuite WARNING] Failed to open SpaceMouse deviceConsider killing other processes that may be using the device such as 3DconnexionHelper (killall 3DconnexionHelper) (spacemouse.py:136)
Traceback (most recent call last):
  File "/home/ps/Desktop/lby_workspace/robocasa/robocasa/scripts/collect_demos.py", line 488, in <module>
    device = SpaceMouse(
  File "/home/ps/Desktop/lby_workspace/robosuite/robosuite/devices/spacemouse.py", line 134, in __init__
    self.device.open(self.vendor_id, self.product_id)  # SpaceMouse
  File "hidraw.pyx", line 143, in hid.device.open
OSError: open failed
 ```
I notice that single_arm_env has been deleted from your code from commit [3f0fab3](https://github.com/ARISE-Initiative/robosuite/commit/3f0fab30faa4dcd85debd0886878fd60fb82b817). I think there are some version problems, can you fix them or give me some suggestions? Any quick reply will be very appreciated.

snasiriany, 2024-12-11:
Do you have a [spacemouse device](https://3dconnexion.com/dk/product/spacemouse-compact/) for teleoperation? If not, you can use keyboard control by adding the flag `--device keyboard` to the command.

The code should work as long as you are using the latest version of robosuite on master branch: https://github.com/ARISE-Initiative/robosuite/tree/master

And the latest commit on robocasa main branch: https://github.com/robocasa/robocasa/tree/main

If you encounter any issues, please post a new issue with a stack trace and we will look into it!

derektan95, 2025-06-20:
Hello @snasiriany, 
Thanks for offering this simulator open-source, and for offering support for cross-embodiment. I wonder if you managed to extend this simulator to other robots (other than Franka and GR1), for example, Unitree's B1? I do see them here: 
https://github.com/ARISE-Initiative/robosuite/tree/robocasa_v0.1/robosuite/models/assets/robots. 

Thanks!