robocasa/robocasa #22: issue still exists: PnPCounterToMicro file signature not found

fortyMiles, 2024-06-17:
hello, I downloaded and rewrited the file. But this error still exits. 

Thanks for your help. 

[image]

snasiriany, 2024-06-17:
Can you verify the latest commit you are on starts with hash id `1370b9e0f747d8` with the message `Fix broken dataset link in dataset_registry.py`?

Also, what's the size of your dataset? You can check in the robocasa folder by running
```
du -h datasets/v0.1/single_stage/kitchen_pnp/PnPCounterToMicrowave/2024-04-27/demo_gentex_im128_randcams.hdf5
```

I'm getting 816 Mb, the previous broken version was only a few kilobytes.

fortyMiles, 2024-06-18:
Wow, thanks! It works, I need to pull the repo. :) 

Hope you have a good day, thank you very much.

closed 2024-06-18 by fortyMiles