robocasa/robocasa #23: How to render the robot trajectory and environment of the generated task in an off-screen environment?

Winniechen2002, 2024-06-21:
In an off-screen environment without X-11, the lack of GLSW and keyboard interaction brings up some errors when running 'python -m robocasa.demos.demo_tasks'.

I have read the code you provided in 'headless issue' but I don't think it might be helpful to this problem. When using the environment for tasks, I found that nothing will be rendered when I don't use mjviewer.

Could you please help me with how to render the task in the server environment?

Winniechen2002, 2024-06-21:
Another question is where's the part for LLM? I thought there would be parts of code using LLM to generate tasks.

GuanxingLu, 2024-08-12:
> In an off-screen environment without X-11, the lack of GLFW and keyboard interaction brings up some errors when running 'python -m robocasa.demos.demo_tasks'.
> 
> I have read the code you provided in 'headless issue' but I don't think it might be helpful to this problem. When using the environment for tasks, I found that nothing will be rendered when I don't use mjviewer.
> 
> Could you please help me with how to render the task in the server environment?

Hi, I found a solution: 

Replacing
https://github.com/robocasa/robocasa/blob/ea0e58e6be42367d2a40085009a3d09a7f104c04/robocasa/scripts/playback_dataset.py#L100
with
```
video_img.append(env.sim.render(mode="offscreen", height=512, width=512, camera_name=cam_name))
```
works for me.

Hope this helps.

snasiriany, 2024-08-15:
This feature is now available. You can run offscreen rendering of the demo task script as:
```
python -m robocasa.demos.demo_tasks --render_offscreen
```

The playback_dataset.py script can also be used to render demos. See this page for additional details: https://robocasa.ai/docs/datasets/inspecting.html

LLMs were used to generate task blueprints and we proceeded to code up the tasks based on the generated blueprints. The LLM prompts are discussed in the paper.

closed 2024-08-15 by snasiriany