robocasa/robocasa #24: a question about action space

mrvgao, 2024-06-21:
Hello, @snasiriany 

Thanks for your previous help. 

I can run robomimic successfully after your guidance. After I train the model, I want to explore the mechanism of policies' outputs. 

I want to find out how to "actions" apply robots kinematics but failed. 

Would you please give some information or source code citations that I can loop these things up? 

Thank you very much.

snasiriany, 2024-08-07:
Apologies for the delayed response! By default, for the datasets we provide, the action space consists of the elements controlling for the arm (Cartesian control, 6 dof), followed by elements controlling the gripper (1 dof), followed by elements controlling the base (4 dof), followed by one (binary) element controlling whether to control the arm or base (1 dof). So 12 dimensions total.

closed 2024-08-07 by snasiriany

abhishek47kashyap, 2026-04-19:
> controlling the base (4 dof)

It's a bit unclear what do those 4 DOF stand for. Could you please clarify? If they were 3 I might have guessed `(x, y, yaw)` but the extra 4th dimension is throwing me off.