robocasa/robocasa #25: Issue about loading similar scene using pybullet

fujiawei0724, 2024-06-26:
Thanks for the authors about the great work, it paves a new way for robot learning. Could you please help me figure out whether I can load similar scenarios in robocasa to pybullet (using the urdf or XML files you provided)? If this is possible, could you please tell me about the loading API of the scenarios in robocasa? Thank you very much.

fujiawei0724, 2024-06-26:
In other word, how a scenario in robocase define the position (and location) of every urdf or XML object in it? Thanks for your reply.

snasiriany, 2024-08-04:
Hi @fujiawei0724 apologies for the late reply. robocasa uses the MuJoCo physics engine which primarily uses the MJCF model format rather than URDF. However, there are some tools to convert MJCF to URDF, like this: https://github.com/iory/mjcf2urdf. Not sure how well this works works, but hopefully it is be useful.

To get the MJCF for a particular environment instance, you can use something like `env.model.get_xml()`, and then pass that into the MJCF -> URDF tool. To get the low-level state in MuJoCo containing all the current positions of individual geoms and joints in the model at any particular time, you can access this via `np.array(env.sim.get_state().flatten())`

Hope this helps!

closed 2024-08-04 by snasiriany