robocasa/robocasa #27: How long do you take to train a task with BC Transformer?

luccachiang, 2024-06-28:
I am using your policy learning code to train a single-task policy. However, training one policy takes several days, which seems a bit long for a typical BC process. I am using single-task Human-50 as the dataset, and training on one 3090 GPU.

How long does it take to train a single task with BC? What kind of GPU do you use to get the result of Figure 7 in your paper? Could you give some advice on reducing training time? Thanks in advance for your response!

amandlek, 2024-07-28:
A 3090 GPU should be more than sufficient, but 3 days seems a little slow to me. Can you please post your robomimic training config here, so we can take a closer look?

cc @snasiriany for vis

snasiriany, 2024-07-30:
I think the numbers you shared here are roughly in the right ballpark. The training time by itself is around 1 day. Most of the time here is being spent on running evals. Do you need to eval every 20 epochs? I typically eval every 100 epochs, and only on a couple tasks. If I'm evaluating on all tasks, I typically wait until training has fully finished and then run the whole eval suite.

closed 2024-07-30 by snasiriany