robocasa/robocasa #31: Is robocasa compatible with tasks in robosuite?

seagul1, 2024-07-08:
when I use 
`robosuite.make(
            env_name="Lift",
            robots="Panda",
            controller_configs=load_controller_config(default_controller="OSC_POSE"),
            # rendering setting
            has_renderer=False,
            has_offscreen_renderer=True,
            use_camera_obs=True,
            camera_depths=True,
            camera_names=["robot0_agentview_left", "robot0_agentview_right", "robot0_agentview_center","robot0_eye_in_hand"],
        )`
I can successfully create the lift env, but when I change the task to "Stack", An error will be reported that 
**AssertionError: Error: Expected one single-armed robot! Got <class 'robosuite.robots.fixed_base_robot.FixedBaseRobot'> type instead.** 
And I tried other single arm task, only Lift and NutAssembly can be created.
Did I mistake something? What should I do to create other task such as PickPlace and Stack?

snasiriany, 2024-07-11:
Thanks for raising this! We are aware of this issue. We need to do some further refactoring of the existing RoboSuite environments. We have mostly fixed this internally, and will push some changes publicly within a week or two.

snasiriany, 2024-08-16:
We just pushed out the latest changes onto the robosuite repo (`robocasa_v0.1` branch): https://github.com/ARISE-Initiative/robosuite/tree/robocasa_v0.1

The single arm envs from before (Lift, Door, PickPlace, Wipe, etc) should all work successfully.

We will roll out updates for two arm envs in the near future.

closed 2024-08-16 by snasiriany