robocasa/robocasa #32: installation on headless ubuntu server

eric-li-2, 2024-07-09:
When trying to install robocasa on an ec2 instance, I get the following errors when attempting to download kitchen assets. Is there a fix for this or is headless just not supported at the moment.

`(robocasa) ubuntu@ip-XXX:~/mlProjects/robocasa$ python robocasa/scripts/download_kitchen_assets.py 
[robosuite WARNING] No private macro file found! (macros.py:53)
[robosuite WARNING] It is recommended to use a private macro file (macros.py:54)
[robosuite WARNING] To setup, run: python /home/ubuntu/mlProjects/robosuite/robosuite/scripts/setup_macros.py (macros.py:55)
Traceback (most recent call last):
  File "/home/ubuntu/mlProjects/robocasa/robocasa/scripts/download_kitchen_assets.py", line 8, in <module>
    import robocasa
  File "/home/ubuntu/mlProjects/robocasa/robocasa/__init__.py", line 1, in <module>
    from robosuite.environments.base import make
  File "/home/ubuntu/mlProjects/robosuite/robosuite/__init__.py", line 1, in <module>
    from robosuite.environments.base import make
  File "/home/ubuntu/mlProjects/robosuite/robosuite/environments/__init__.py", line 1, in <module>
    from .base import REGISTERED_ENVS, MujocoEnv
  File "/home/ubuntu/mlProjects/robosuite/robosuite/environments/base.py", line 13, in <module>
    from robosuite.utils.binding_utils import MjRenderContextOffscreen, MjSim
  File "/home/ubuntu/mlProjects/robosuite/robosuite/utils/binding_utils.py", line 12, in <module>
    import mujoco
  File "/home/ubuntu/anaconda3/envs/robocasa/lib/python3.9/site-packages/mujoco/__init__.py", line 47, in <module>
    from mujoco.gl_context import *
  File "/home/ubuntu/anaconda3/envs/robocasa/lib/python3.9/site-packages/mujoco/gl_context.py", line 41, in <module>
    from mujoco.egl import GLContext as _GLContext
  File "/home/ubuntu/anaconda3/envs/robocasa/lib/python3.9/site-packages/mujoco/egl/__init__.py", line 31, in <module>
    from mujoco.egl import egl_ext as EGL
  File "/home/ubuntu/anaconda3/envs/robocasa/lib/python3.9/site-packages/mujoco/egl/egl_ext.py", line 27, in <module>
    from OpenGL import EGL
  File "/home/ubuntu/anaconda3/envs/robocasa/lib/python3.9/site-packages/OpenGL/EGL/__init__.py", line 2, in <module>
    from OpenGL.raw.EGL._types import *
  File "/home/ubuntu/anaconda3/envs/robocasa/lib/python3.9/site-packages/OpenGL/raw/EGL/_types.py", line 87, in <module>
    raw_eglQueryString = _p.PLATFORM.EGL.eglQueryString
AttributeError: 'NoneType' object has no attribute 'eglQueryString'`

snasiriany, 2024-07-11:
If you are only trying to download the assets, but not run any simulations, I think there's a way to circumvent the issue where you disable the chain of imports that rely on mujoco. Since the process of downloading assets does not rely on mujoco per se.

More broadly, I think this is a larger issue to do with mujoco setup, and I think some folks have discussed this, eg. here: https://github.com/google-deepmind/dm_control/issues/136

Does following the suggested solution in that link solve the issue?

eric-li-2, 2024-07-11:
I got it working but I'm not sure how. I think the package that worked was "sudo apt install libglew-dev" but there were probably a couple other graphics packages I installed before.

snasiriany, 2024-07-11:
Great, thanks for letting us know!

closed 2024-07-11 by snasiriany

LukeLIN-web, 2025-05-23:
Same problem, any advice to run simulation like https://github.com/Abhiram824/openpi/blob/main/examples/robocasa/main.py#L195  
in headless server? I can download kitchen assets , but don't have GUI.