robocasa/robocasa #33: Issue about dataset

mrvgao, 2024-07-09:
Hello, 

When I am using the robocasa to train a policy, I can only find the dataset "demo.hdf5" and "demo_gentex_im128_randoms.hdf5". But based on the paper, there is a bigger dataset called "generated-3000". Would you please tell me how can I find this dataset? 

Thank you so much !

snasiriany, 2024-07-11:
We are still finding ways to publicly release the generated datasets in a practical way, since the file sizes are large. Please stay tuned on that front!

mrvgao, 2024-07-15:
Thank you! If the file is so large, I think maybe you can build a server and use some token to verify the accessibility. Like LLAMA, users, like me, can apply the access code, and then use this to wget directly from your server.

Abhiram824, 2024-07-28:
Hello, the datasets are uploaded now. Please pull the latest changes from the Robocasa branch and run download_datasets.py to access them