robocasa/robocasa #38: Kitchen layout and style ids for demonstration datasets

sunwookim1214, 2024-07-15:
Is there an organized table showing each demo's kitchen layout and style ids?

abhihjoshi, 2024-07-20:
Do you mind elaborating what you mean by this? I believe each demo can be run in any kitchen layout using any style. If you are referring to a list of layouts and styles, feel free to refer to the RoboCasa [paper](https://robocasa.ai/assets/robocasa_rss24.pdf). Figure 3 lists all the kitchen layouts in order from top to bottom and left to right.

snasiriany, 2024-07-22:
We enumerate the kitchen scenes (layouts and styles) [here](https://robocasa.ai/docs/tasks_scenes_assets/kitchen_scenes.html).

Additionally, the layout id (number) to name mapping is [here](https://github.com/robocasa/robocasa/blob/130e919f2fec5e4aa929ac7376126093d1ce32c8/robocasa/utils/kitchen_utils.py#L2-L15) and for styles it is [here](https://github.com/robocasa/robocasa/blob/130e919f2fec5e4aa929ac7376126093d1ce32c8/robocasa/models/arenas/layout_builder.py#L66-L79).

closed 2024-07-22 by snasiriany

sunwookim1214, 2024-07-23:
Perhaps I am not yet familiar enough with the code, but I'm a bit lost on how I can set the layout and style for each demo.

I would greatly appreciate it if you could elaborate on how I might be able to do so!

abhihjoshi, 2024-07-23:
When running the demos, you can pass in the layout and the style as arguments. For example to run the `demo_kitchens.py` file with layout 3 (L_shaped_large_arena) and with style 8 (rustic), you might run the command:

`python robocasa/demos/demo_kitchens.py --task {TASK} --layout 3 --style 8`

You can do the same with `demo_teleop.py` as well. `demo_objects.py` is meant to just visualize objects and `demo_tasks.py` replays previously collected demonstrations where the layout and styles were already set when collecting those demonstrations. As a result, we do not include a layout and style argument for these demos.

sunwookim1214, 2024-07-24:
Is there a way I can select specific demos in demo_tasks.py according to layout, styles, and objects in the scene, or are they not yet categorized as such?

FYI I'm trying to use RoboCasa to collect data for my scene change detection framework, so it would be nice if I could specify the scenes to be played.

abhihjoshi, 2024-07-24:
I would recommend taking a look at `robocasa/scripts/playback_demonstration.py` to see how to read datasets (and information regarding these datasets). You can read datasets using the h5py module. Line 299 shows you how to use h5py to open and read a dataset hdf5 file. Specifically, if you look at lines 358-361, you'll see how we actually read from the file. If I not mistaken, the `layout_id` and `style_id` is under the episode meta data (`"ep_meta"`, indexed at line 361). You should be able to iterate through all the episodes in a dataset and check which ones have the layout_id and style_id you are looking for. From there you can choose the episode based on those attributes.
```
states = f["data/{}/states".format(ep)][()]
initial_state = dict(states=states[0])
initial_state["model"] = f["data/{}".format(ep)].attrs["model_file"]
initial_state["ep_meta"] = f["data/{}".format(ep)].attrs.get("ep_meta", None)
```

sunwookim1214, 2024-08-05:
Thank you for your help!

I was wondering, how many demonstration datasets (variations in kitchen layout, style, and object placement) are there for each of the 30 uploaded demonstration tasks (In the data registry script there are 25 atomic tasks and 5 composite tasks)?

sunwookim1214, 2024-08-06:
Also, what is the role of the 'filter_keys' variable in the demo_tasks script? It seems like changing the variable doesn't affect anything about which demonstration plays, only the task type and demo number does.

sunwookim1214, 2024-08-08:
I would appreciate it if you could help me out as soon as possible! I have a deadline for my paper :)

snasiriany, 2024-08-09:
Re: datasets. There are 30 human datasets (25 atomic, 5 composite), and 24 mimicgen datasets (24 atomic, excludes navigation task).

Re: filter_key. The filter_key features serves as a way to reference subsets of the datasets (eg. 10 random demos, 20 random demos, etc). If you want to train on fewer demos than the full datasets, you can use this feature. You can run `python robocasa/scripts/get_dataset_info.py --dataset <ds-path>` to see a list of available filter keys.

sunwookim1214, 2024-08-12:
Thank you for your reply.

I just wanted to also make sure, am I correct in identifying that there are 55 demos per dataset (task)? 

Also, would greatly appreciate it if I could know the composition of kitchen layout and styles used in the demos, per task. 

Thank you so much!

snasiriany, 2024-08-13:
Yes, 55 demos per task! Some may have slightly fewer due to the script that dropped demos. You can use the `50_demos` filter key to use 50 demos, we have that guaranteed number per dataset.

To see the composition, run `python robomimic/scripts/get_dataset_info.py --dataset <ds-path> --get_ep_meta_stats`