robocasa/robocasa #39: Do robocasa support customized task/scene design?

YY-GX, 2024-07-18:
Hi, thank you for providing this awesome simulation environment! I'm wondering whether the robocasa could support customize the tasks by myself?

abhihjoshi, 2024-07-20:
Yes, you can create your own tasks in RoboCasa. I would recommend using some of the already written tasks as a guideline when creating a new task. Add your new task to either the `robocasa/environments/kitchen/multi_stage` or `robocasa/environments/kitchen/single_stage` (dependent on what your task is). You should also add your task to `robocasa/__init__.py`.

snasiriany, 2024-07-22:
Adding to the above, you can also create new scene layouts in [this folder](https://github.com/robocasa/robocasa/tree/130e919f2fec5e4aa929ac7376126093d1ce32c8/robocasa/models/assets/kitchen_layouts), and new styles in [this folder](https://github.com/robocasa/robocasa/tree/130e919f2fec5e4aa929ac7376126093d1ce32c8/robocasa/models/assets/kitchen_layouts/styles). I would recommend checking out the existing example layouts/styles for guidance.

New layouts need to be registered [here](https://github.com/robocasa/robocasa/blob/130e919f2fec5e4aa929ac7376126093d1ce32c8/robocasa/utils/kitchen_utils.py#L2-L15) and new styles need to be registered [here](https://github.com/robocasa/robocasa/blob/130e919f2fec5e4aa929ac7376126093d1ce32c8/robocasa/models/arenas/layout_builder.py#L66-L79).

closed 2024-07-22 by snasiriany