robocasa/robocasa #40: Data Generation with MimicGen

Guiliang, 2024-07-20:
Hi,

Is there a plan to release the data generation script based on MimicGen? Additionally, if we generate data directly using [MimicGen](https://mimicgen.github.io/), can we achieve performance that aligns with the results reported in the paper?

snasiriany, 2024-07-22:
We will be releasing our MimicGen datasets shortly. The generation scripts will also come later, most likely towards the end of summer, after we do some cleanup and consolidate the changes with the MimicGen development team.

closed 2024-07-22 by snasiriany

ZzZZCHS, 2024-08-23:
Hi @snasiriany,

I noticed that MimicGen released their generation code last month. Could I use that code directly for the RoboCasa dataset, or do you plan to release specific generation code or instructions for RoboCasa?

Thank you!

snasiriany, 2024-08-23:
@ZzZZCHS We will need to release some specific data generation code that defines all of the stages and objects to track. It's still in progress. Please stay tuned for a few more weeks as we iron out a few things. In the meanwhile, the datasets we already generated are already released.

snasiriany, 2024-12-12:
fyi: mimicgen support now added, please see here: https://github.com/robocasa/robocasa/issues/108