robocasa/robocasa #42: Is it possible to reload a scene exactly as it was previously generated?

karthikm-0, 2024-07-25:
Thank you for this great framework! I had a question about saving scene information from randomly generated scenes. I think ep_meta contains this information (inside object_cfgs). I was wondering if there is a mechanism to reload this data when creating the scene? I know layouts and styles can be provided as inputs but I want the scene to be identical.

amandlek, 2024-07-28:
I believe you should be able to reload scenes by reloading the mujoco xml saved in a demonstration. This is often done in scripts such as [this one](https://github.com/ARISE-Initiative/robomimic/blob/8c23c696260ffb2709b95cd868cf3fc6480afeba/robomimic/scripts/playback_dataset.py#L123) in robomimic. 

I believe this should be sufficient, but please let us know if this doesn't solve your issue. 

Also tagging @snasiriany for vis.

closed 2024-07-28 by amandlek

karthikm-0, 2024-07-30:
Thanks for this reference @amandlek! However, when setting the state, the items in the task still change color and texture.

amandlek, 2024-07-30:
@snasiriany do we save the generated colors and textures in the model xmls?

@karthikm-0 to be clear, this is even after resetting the model xml?

snasiriany, 2024-07-30:
@karthikm-0 can you share some code with us so we have some more context for what you are trying?

karthikm-0, 2024-08-01:
Here's some code for getting the state:

`
    if args.dataset_path is not None:
        f = h5py.File(args.dataset_path, "r")
        demos_data = f['data']
        demos = list(f['data'].keys())
        ep = demos[args.episode_num] if args.episode_num is not None else demos[0]
        demo_data = demos_data[ep]
        obs = demo_data['obs']['robot0_agentview_left_image']
        states = f["data/{}/states".format(ep)][()]
        initial_state = dict(states=states[0])
        # initial_state["model"] = f["data/{}".format(ep)].attrs["model_file"]
        logging.info("Loaded episode: {}".format(ep))
        # Save the last image
        imageio.imsave("/home/karthikm/photomanipulator-sim/octo/goal_image.png", obs[-1])

    else:
        initial_state = None
`

And the reset function is essentially the same code from the playback_dataset script:

`def reset_to(env, state):
    """
    Reset to a specific simulator state.

    Args:
        state (dict): current simulator state that contains one or more of:
            - states (np.ndarray): initial state of the mujoco environment
            - model (str): mujoco scene xml
    
    Returns:
        observation (dict): observation dictionary after setting the simulator state (only
            if "states" is in @state)
    """
    should_ret = False
    if "model" in state:
        if state.get("ep_meta", None) is not None:
            # set relevant episode information
            ep_meta = json.loads(state["ep_meta"])
        else:
            ep_meta = {}
        if hasattr(env, "set_attrs_from_ep_meta"): # older versions had this function
            env.set_attrs_from_ep_meta(ep_meta)
        elif hasattr(env, "set_ep_meta"): # newer versions
            env.set_ep_meta(ep_meta)
        # this reset is necessary.
        # while the call to env.reset_from_xml_string does call reset,
        # that is only a "soft" reset that doesn't actually reload the model.
        env.reset()
        robosuite_version_id = int(robosuite.__version__.split(".")[1])
        if robosuite_version_id <= 3:
            from robosuite.utils.mjcf_utils import postprocess_model_xml
            xml = postprocess_model_xml(state["model"])
        else:
            # v1.4 and above use the class-based edit_model_xml function
            xml = env.edit_model_xml(state["model"])

        env.reset_from_xml_string(xml)
        env.sim.reset()
        # hide teleop visualization after restoring from model
        # env.sim.model.site_rgba[env.eef_site_id] = np.array([0., 0., 0., 0.])
        # env.sim.model.site_rgba[env.eef_cylinder_id] = np.array([0., 0., 0., 0.])
        if hasattr(env, "unset_ep_meta"): # unset the ep meta after reset complete
            env.unset_ep_meta()
    if "states" in state:
        env.sim.set_state_from_flattened(state["states"])
        env.sim.forward()
        should_ret = True

    # update state as needed
    if hasattr(env, "update_sites"):
        # older versions of environment had update_sites function
        env.update_sites()
    if hasattr(env, "update_state"):
        # later versions renamed this to update_state
        env.update_state()

    # if should_ret:
    #     # only return obs if we've done a forward call - otherwise the observations will be garbage
    #     return get_observation()
    return None
`

amandlek, 2024-08-01:
If you look at the link in my original reply (e.g. [here](https://github.com/ARISE-Initiative/robomimic/blob/8c23c696260ffb2709b95cd868cf3fc6480afeba/robomimic/scripts/playback_dataset.py#L123)) it will show you how to reset to a start state using the model xml. In particular, [this line](https://github.com/ARISE-Initiative/robomimic/blob/4690f3ecfa658e1848b2e3fcf8dc2db350f9ffb9/robomimic/scripts/playback_dataset.py#L283) shows you how to read the model xml from the file, and then you can pass that dictionary containing the "states" and "model" key into the env.reset_to function. 

I'm also a little confused why you aren't just directly following robomimic for creating the environment, and instead copying methods around manually (such as the reset_to method).

karthikm-0, 2024-08-01:
Thanks @amandlek! I must have missed this line in the script. Re: not using robomimic - I'm very new to this codebase. But for my use case of inferencing a trained policy, I assumed I would need to load a robosuite env (not robomimic) and set up the environment in a specific way. Since robosuite doesn't have a reset_to function, I am using it locally. Is this the wrong approach?

amandlek, 2024-08-01:
No you can directly use robomimic for policy evaluation. We have a policy evaluation script - for example [here[(https://robomimic.github.io/docs/tutorials/using_pretrained_models.html)

brunomachado37, 2024-08-23:
The `robomimic/scripts/playback_dataset.py` script actually renders random kitchen styles when replaying a trajectory from the dataset. Here it is an example of a trajectory re-played from the actions and the dataset observations:

https://github.com/user-attachments/assets/39575921-fc7a-4bd6-8ea8-40eb37dec338

https://github.com/user-attachments/assets/029497bd-6b01-4209-aee9-af85df292942

Is there a way of replaying a trajectory with the exact same kitchen style as in the original dataset observations?