robocasa/robocasa #46: How to add more robots in one environment using robocasa?

ypz21, 2024-08-02:
Hello,
Thanks for building this amazing project for manipulation tasks in kitchen!
However, my project requires multi-arm robot or more robots cooperation in one environment, and I would like to inquire that if there is a convenient way to add more robots in one environment?

snasiriany, 2024-08-06:
Yes this is possible! RoboCasa is built on top of robosuite, and robosuite has a few multi-robot envs for reference, eg TwoArmTransport: https://github.com/ARISE-Initiative/robosuite/blob/master/robosuite/environments/manipulation/two_arm_transport.py

To include multiple robots, you'd need to specify a list of robots [here](https://github.com/robocasa/robocasa/blob/c3e89f7f93d44b2e0c016ba6837a655640e1f5b7/robocasa/environments/kitchen/kitchen.py#L41), eg. ["PandaMobile", "PandaMobile"]. And then modify some of the code to handle custom logic for your multi-robot setup accordingly. One place to consider is the initialization of the robot base positions (see [here](https://github.com/robocasa/robocasa/blob/c3e89f7f93d44b2e0c016ba6837a655640e1f5b7/robocasa/environments/kitchen/kitchen.py#L259-L262)).

Hope this helps, and feel free to open the issue again if you have followup questions!

ypz21, 2024-08-08:
Hi there!
Follow your advice I successfully added another PandaMobile in the kitchen arena, and this is the current kitchen scene:

[image]

Moreover, I added the controller and camera switching rules, which are the same logic as Robosite demo_device_control.py The result shows that the controller could be successfully toggled between two agents, but the camera can not.

I also tried to set different camera_id in ```env.viewer.set_camera()``` for a different rendering view, but the perspective remains the same. I am wondering which part of the Robocasa codes should I modify to resolve this issue? Any suggestion or help is appreciated!

Robosuite's logic for toggling camera and controller: 
```
            # If the current grasp is active (1) and last grasp is not (-1) (i.e.: grasping input just pressed),
            # toggle arm control and / or camera viewing angle if requested
            if last_grasp < 0 < grasp:
                if args.switch_on_grasp:
                    args.arm = "left" if args.arm == "right" else "right"
                if args.toggle_camera_on_grasp:
                    cam_id = (cam_id + 1) % num_cam
                    env.viewer.set_camera(camera_id=cam_id)
```

My code for toggling camera and controller: 
```
        # Change the active robot based on grasp infomation if there are two robots
        if last_grasp < 0 < grasp and env.num_robots == 2:
            if active_robot == env.robots[0]: # change active robot to robot1
                active_robot = env.robots[1]
                # camera_id = (camera_id + 1) % num_cam
                # print(camera_id)
                # print(env.sim.model.camera_names[camera_id])
                camera_id = env.sim.model.camera_name2id('robot1_robotview')
                env.viewer.set_camera(camera_id=camera_id)
            elif active_robot == env.robots[1]: # change active robot to robot0
                active_robot = env.robots[0]
                camera_id = env.sim.model.camera_name2id('robot0_robotview')
                env.viewer.set_camera(camera_id=camera_id)
            else:
                raise ValueError("Invalid active robot")
```

ypz21, 2024-08-09:
Dear Soroush,

Follow your advice I successfully added another PandaMobile in the kitchen arena. And I have added some comments on the closed issue.

Moreover, I added the controller and camera switching rules, which are the same logic as Robosite demo_device_control.py The result shows that the controller could be successfully toggled between two agents, but the camera can not.

I also tried to set different camera_id in ```env.viewer.set_camera()``` for a different rendering view, but the perspective remains the same. I am wondering which part of the Robocasa codes should I modify to resolve this issue? Any suggestion or help is appreciated!

Robosuite's logic for toggling camera and controller: 
```
            # If the current grasp is active (1) and last grasp is not (-1) (i.e.: grasping input just pressed),
            # toggle arm control and / or camera viewing angle if requested
            if last_grasp < 0 < grasp:
                if args.switch_on_grasp:
                    args.arm = "left" if args.arm == "right" else "right"
                if args.toggle_camera_on_grasp:
                    cam_id = (cam_id + 1) % num_cam
                    env.viewer.set_camera(camera_id=cam_id)
```

My code for toggling camera and controller: 
```
        # Change the active robot based on grasp infomation if there are two robots
        if last_grasp < 0 < grasp and env.num_robots == 2:
            if active_robot == env.robots[0]: # change active robot to robot1
                active_robot = env.robots[1]
                # camera_id = (camera_id + 1) % num_cam
                # print(camera_id)
                # print(env.sim.model.camera_names[camera_id])
                camera_id = env.sim.model.camera_name2id('robot1_robotview')
                env.viewer.set_camera(camera_id=camera_id)
            elif active_robot == env.robots[1]: # change active robot to robot0
                active_robot = env.robots[0]
                camera_id = env.sim.model.camera_name2id('robot0_robotview')
                env.viewer.set_camera(camera_id=camera_id)
            else:
                raise ValueError("Invalid active robot")
```

Sincerely,

Puzhen Yuan

ypz21, 2024-08-09:
Oh I just discovered that change the renderer from 'mjviewer' back to 'mujoco' fix the problem.
I don't know what's wrong with 'mjviewer'...

snasiriany, 2024-08-09:
pressing `[` or `]` on the mjviewer should toggle among all cameras, and as long as the camera names for robot0 and robot1 are distinct from one another you should be able to toggle between them successfully. can you try this?

ypz21, 2024-08-09:
Yes, I have tried pressing ```[``` and ```]``` on mjviewer and it works! Thanks for answering!

Now my ```env.sim.model.camera_names``` looks like this:
```
('robot0_robotview', 'robot0_agentview_center', 'robot0_agentview_left', 'robot0_agentview_right', 'robot0_frontview', 'robot0_eye_in_hand', 'robot1_robotview', 'robot1_eye_in_hand')
```
It seems that these robot0 cameras are from ```kitchen_utils.CAM_CONFIGS```? After adding another robot, only ```'robot1_robotview'```, ```'robot1_eye_in_hand'``` are automacically added. Could you please tell me which part of the codes should I modify to automatically add cameras like ```robot1_frontview``` or ```robot1_agentview_center```?