robocasa/robocasa #48: Do robocasa support customized new items?

hahahalove57, 2024-08-12:
Hi, thank you for providing this awesome simulation environment! I'm wondering whether the robocasa could support customize new objects by myself?
And I customize one item on the counter_main,but it suspends in the air. I wanna know how its realized for the z level of one item on the counter?
[image]
[image]
[image]

snasiriany, 2024-08-12:
Hi @hahahalove57 are these new objects that you ported into RoboCasa? If you can provide some additional context (model files), code, how you ported objects in, what modifications can be made, we can try to help!

hahahalove57, 2024-08-13:
> Hi @hahahalove57 are these new objects that you ported into RoboCasa? If you can provide some additional context (model files), code, how you ported objects in, what modifications can be made, we can try to help!

Hello! Thank you for your reply. I modeled the definition of knife_blocks within robocasa and added it to the counter_accessory of a scenario with the help of the code about cakes provided within the objarverse. I noticed that the xml files for the rest of the objects define scale as [1 1 1], and that changing this value (e.g. [0.81 0.81 0.81 0.81] in the xml for the knife_block) causes it to shrink in size in the scene and appear to be hovering. Does this physics simulation take gravity into account? I read the code that specifies the value of the object's z-axis and found that in line 323 of placement_sample.py, object_z -= obj.bottom_offset[-1]. But I don't really understand how this bottom_offset[-1] is defined, is it related to the enclosing box defined in the xml? Since I don't fully grasp the details of the xml file's definition of the object, I hope you can help me with my confusion！

hahahalove57, 2024-08-13:
<mujoco model="cake_25">
  <asset>
    <mesh file="visual/model_normalized_0.obj" name="model_normalized_0_vis" scale="1 1 1" refquat="-1.0 0.0 0.0 -2.220446049250313e-16"/>
    <mesh file="collision/model_normalized_collision_22.obj" name="model_normalized_collision_22._coll" scale="1 1 1" refquat="-1.0 0.0 0.0 -2.220446049250313e-16"/>
    <mesh file="collision/model_normalized_collision_23.obj" name="model_normalized_collision_23._coll" scale="1 1 1" refquat="-1.0 0.0 0.0 -2.220446049250313e-16"/>
    <mesh file="collision/model_normalized_collision_21.obj" name="model_normalized_collision_21._coll" scale="1 1 1" refquat="-1.0 0.0 0.0 -2.220446049250313e-16"/>
    <mesh file="collision/model_normalized_collision_20.obj" name="model_normalized_collision_20._coll" scale="1 1 1" refquat="-1.0 0.0 0.0 -2.220446049250313e-16"/>
    <mesh file="collision/model_normalized_collision_24.obj" name="model_normalized_collision_24._coll" scale="1 1 1" refquat="-1.0 0.0 0.0 -2.220446049250313e-16"/>
    <mesh file="collision/model_normalized_collision_30.obj" name="model_normalized_collision_30._coll" scale="1 1 1" refquat="-1.0 0.0 0.0 -2.220446049250313e-16"/>
    <mesh file="collision/model_normalized_collision_18.obj" name="model_normalized_collision_18._coll" scale="1 1 1" refquat="-1.0 0.0 0.0 -2.220446049250313e-16"/>
    <mesh file="collision/model_normalized_collision_19.obj" name="model_normalized_collision_19._coll" scale="1 1 1" refquat="-1.0 0.0 0.0 -2.220446049250313e-16"/>
    <mesh file="collision/model_normalized_collision_31.obj" name="model_normalized_collision_31._coll" scale="1 1 1" refquat="-1.0 0.0 0.0 -2.220446049250313e-16"/>
    <mesh file="collision/model_normalized_collision_25.obj" name="model_normalized_collision_25._coll" scale="1 1 1" refquat="-1.0 0.0 0.0 -2.220446049250313e-16"/>
    <mesh file="collision/model_normalized_collision_27.obj" name="model_normalized_collision_27._coll" scale="1 1 1" refquat="-1.0 0.0 0.0 -2.220446049250313e-16"/>
    <mesh file="collision/model_normalized_collision_26.obj" name="model_normalized_collision_26._coll" scale="1 1 1" refquat="-1.0 0.0 0.0 -2.220446049250313e-16"/>
    <mesh file="collision/model_normalized_collision_9.obj" name="model_normalized_collision_9._coll" scale="1 1 1" refquat="-1.0 0.0 0.0 -2.220446049250313e-16"/>
    <mesh file="collision/model_normalized_collision_8.obj" name="model_normalized_collision_8._coll" scale="1 1 1" refquat="-1.0 0.0 0.0 -2.220446049250313e-16"/>
    <mesh file="collision/model_normalized_collision_6.obj" name="model_normalized_collision_6._coll" scale="1 1 1" refquat="-1.0 0.0 0.0 -2.220446049250313e-16"/>
    <mesh file="collision/model_normalized_collision_7.obj" name="model_normalized_collision_7._coll" scale="1 1 1" refquat="-1.0 0.0 0.0 -2.220446049250313e-16"/>
    <mesh file="collision/model_normalized_collision_5.obj" name="model_normalized_collision_5._coll" scale="1 1 1" refquat="-1.0 0.0 0.0 -2.220446049250313e-16"/>
    <mesh file="collision/model_normalized_collision_4.obj" name="model_normalized_collision_4._coll" scale="1 1 1" refquat="-1.0 0.0 0.0 -2.220446049250313e-16"/>
    <mesh file="collision/model_normalized_collision_0.obj" name="model_normalized_collision_0._coll" scale="1 1 1" refquat="-1.0 0.0 0.0 -2.220446049250313e-16"/>
    <mesh file="collision/model_normalized_collision_1.obj" name="model_normalized_collision_1._coll" scale="1 1 1" refquat="-1.0 0.0 0.0 -2.220446049250313e-16"/>
    <mesh file="collision/model_normalized_collision_3.obj" name="model_normalized_collision_3._coll" scale="1 1 1" refquat="-1.0 0.0 0.0 -2.220446049250313e-16"/>
    <mesh file="collision/model_normalized_collision_2.obj" name="model_normalized_collision_2._coll" scale="1 1 1" refquat="-1.0 0.0 0.0 -2.220446049250313e-16"/>
    <mesh file="collision/model_normalized_collision_17.obj" name="model_normalized_collision_17._coll" scale="1 1 1" refquat="-1.0 0.0 0.0 -2.220446049250313e-16"/>
    <mesh file="collision/model_normalized_collision_16.obj" name="model_normalized_collision_16._coll" scale="1 1 1" refquat="-1.0 0.0 0.0 -2.220446049250313e-16"/>
    <mesh file="collision/model_normalized_collision_28.obj" name="model_normalized_collision_28._coll" scale="1 1 1" refquat="-1.0 0.0 0.0 -2.220446049250313e-16"/>
    <mesh file="collision/model_normalized_collision_14.obj" name="model_normalized_collision_14._coll" scale="1 1 1" refquat="-1.0 0.0 0.0 -2.220446049250313e-16"/>
    <mesh file="collision/model_normalized_collision_15.obj" name="model_normalized_collision_15._coll" scale="1 1 1" refquat="-1.0 0.0 0.0 -2.220446049250313e-16"/>
    <mesh file="collision/model_normalized_collision_29.obj" name="model_normalized_collision_29._coll" scale="1 1 1" refquat="-1.0 0.0 0.0 -2.220446049250313e-16"/>
    <mesh file="collision/model_normalized_collision_11.obj" name="model_normalized_collision_11._coll" scale="1 1 1" refquat="-1.0 0.0 0.0 -2.220446049250313e-16"/>
    <mesh file="collision/model_normalized_collision_10.obj" name="model_normalized_collision_10._coll" scale="1 1 1" refquat="-1.0 0.0 0.0 -2.220446049250313e-16"/>
    <mesh file="collision/model_normalized_collision_12.obj" name="model_normalized_collision_12._coll" scale="1 1 1" refquat="-1.0 0.0 0.0 -2.220446049250313e-16"/>
    <mesh file="collision/model_normalized_collision_13.obj" name="model_normalized_collision_13._coll" scale="1 1 1" refquat="-1.0 0.0 0.0 -2.220446049250313e-16"/>
[... cut ...]
      <site rgba="0 0 0 1" size="0.01" pos="-0.2320324546589323 -0.24096115720684336 -0.5" name="ext_p0"/>
      <site rgba="1 0 0 1" size="0.01" pos="0.2320324546589323 -0.24096115720684336 -0.5" name="ext_px"/>
      <site rgba="0 1 0 1" size="0.01" pos="-0.2320324546589323 0.24096115720684336 -0.5" name="ext_py"/>
      <site rgba="0 0 1 1" size="0.01" pos="-0.2320324546589323 -0.24096115720684336 0.5" name="ext_pz"/>
    </body>
  </worldbody>
</mujoco>

this is the xml file that based o the code of cake_05 from the objarverse

closed 2024-08-17 by hahahalove57