robocasa/robocasa #49: Issues with robocasa dataset downloads and converting states to observations

karthikm-0, 2024-08-13:
Hi,

I noticed a few issues with the datasets. First, I believe the PnpCounterToStove dataset is labelled incorrectly - it is actually PnpMicrowaveToCounter. This is based on downloading the dataset and sampling a few episodes. Is PnpCounterToStove available somewhere else (or can it be made available)?

Second, I have had challenges in extracting the observations from the hdf5 files consistently. For instance, most datasets have 55 demos but using the commands listed here (https://robocasa.ai/docs/use_cases/policy_learning.html) to convert the states to observations often leads to demos fewer than that number - usually around 53-54 but sometimes much fewer. This seems to be random and running the same command several times on the same dataset yields a different numbers of episodes; this command is where this happens:

`OMP_NUM_THREADS=1 MPI_NUM_THREADS=1 MKL_NUM_THREADS=1 OPENBLAS_NUM_THREADS=1 python 
robomimic/scripts/dataset_states_to_obs.py --dataset <ds-path>
`

I also noticed that when extracting the observations using , some episodes result in black pixels for all steps. I only tested this with observations from the robot0_agentview_left camera (with resolution 256x256). Sometimes, I see a speck of pixels (e.g., yellow color) surrounded by black pixels for the rest of the image.  I've attached one example (where after filtering episodes with frames containing pure black I am still left with 2 episodes that are mostly black).

Lastly, some of the downloaded datasets do not include env_info so when converting them using convert_robosuite.py they will yield errors. I remedied this by adding some logic to extract env_kwargs since it contains env_info:

`    # store env meta
    env_name = f["data"].attrs.get("env", None)
    if "env_info" in f["data"].attrs:
        env_info = json.loads(f["data"].attrs["env_info"])
        # switch to using solid robot
        env_info["translucent_robot"] = False
    else:
        env_args = f["data"].attrs.get("env_args", None)
        # Make into dict
        env_args = json.loads(env_args)
        env_info = env_args["env_kwargs"] if env_args is not None else None
        env_info["translucent_robot"] = False
`

[image]

snasiriany, 2024-08-13:
Hi @karthikm-0 thanks a lot for taking the time to share these things with us.

Re: PnPCounterToStove broken links. Great catch! I've fix the registry with the correct url links. Please pull the latest code and re-run ` python robocasa/scripts/download_datasets.py --tasks PnPCounterToStove --overwrite`

Re: extracting all demos. We are aware of this issue. That script runs multiple subprocesses and when running multiple processes at the same time, some of the subprocesses can fail. If you do not want to lose any demos in the process, it's worth running that script with `--num_procs 1` which only runs one process. Let us know if with this the issue goes away.

Also, please let us know if there are specific datasets for which you saw way fewer than 55 demos generated.

Re: black pixels. I wonder if this is because your visual buffer is filling up. Which version of mujoco are you on? You can check with `pip freeze | grep mujoco`. We have been using `mujoco==3.1.1` which hasn't caused issues for us. I know with `3.2.0` and `3.2.1` there have been some issues. Also, what computer specs / OS are you on?

Re: env_info. The `env_args` key, which can be accessed as `env_args = json.loads(f["data"].attrs["env_args"])` contains all the necessary information to play back datasets and extract observations. All datasets should have this key already, but let me know if this is not the case.

snasiriany, 2024-08-20:
@karthikm-0 feel free to re-open the issue if you have followup comments.

closed 2024-08-20 by snasiriany

karthikm-0, 2024-08-26:
Sorry for the delay in my response. Thanks for updating the links, it now works! I was able to get all the data using a single process so that is also great to hear. I believe the problem of fewer demonstrations will not be a concern using this command. 

I'm also using `mujoco==3.1.1` on Ubuntu 20.04 with an Nvidia A6000 + an Intel Xeon W-2295 @ 3.00 GHz.

About the env_args: Yes that is correct, but the codebase assumes env_info is a key in the dataset rather than inside env_args so the logic I added was just to address that because some of the datasets did not include this key (but could still be accessed via env_args).

snasiriany, 2024-08-28:
Interesting note about the partially blank observation images. We will keep an eye out for this and update you if we find something. I believe the datasets we've publicly released have no such issues with the observations.

Re: env_info. Which script uses env_info? I think the only one I found is in robomimic's convert_robosuite.py script, but we only need to run that once when collecting the demos initially, and it's not needed after that.

karthikm-0, 2024-08-28:
I believe it is convert_robosuite but it failed for some of the datasets that I downloaded because they were missing the env_info key; but adding the code above addresses it. If you would like me to investigate, I can run the scripts again to see which ones failed.

snasiriany, 2024-08-29:
convert_robosuite is only meant to run on raw datasets (to turn them into robomimic's dataset format). for datasets that don't have env_info, that means convert_robosuite has already been run.