robocasa/robocasa #51: Render dataset with the Omniverse renderer

brunomachado37, 2024-08-22:
Hello,

First of all, thank you for the amazing open source work.

In the paper you mentioned that _"For the public release we will provide users the option to render all datasets with the Omniverse renderer."_

I couldn't find a script in the code for rendering the dataset trajectories with the Omniverse renderer. Is there a simple way of adapting the `robomimic/scripts/dataset_states_to_obs.py` script to change the renderer?

Also, the code was released under MIT license. I assume the dataset is under the same license, is that correct?

abhihjoshi, 2024-08-22:
> In the paper you mentioned that _"For the public release we will provide users the option to render all datasets with the Omniverse renderer."_

Under the `robocasa/utils/usd` directory, you will find the code required to export MuJoCo scenes such as RoboCasa as USD files (which can then be loaded into Omniverse). However, this code may be outdated. We are currently working on releasing a script to render scenes with Omniverse, and this should be out very soon!

snasiriany, 2024-08-22:
There is a script here for converting dataset episodes to usd files: https://github.com/robocasa/robocasa/blob/main/robocasa/scripts/generate_usd_trajectories.py

Once you have the USD files you can load them in Omniverse and render directly that way. We will make the instructions more clear in our documentation in the future.

The datasets are under the same license, correct.