robocasa/robocasa #53: GymWrapper compatibility issue

lorandcheng, 2024-08-23:
Hello,

I'm trying to wrap robocasa envs using the `GymWrapper` from robosuite. I noticed that robosuite envs don't have all the necessary attributes (`use_object_obs`, `reward_cale`, `reward_shaping`)

These args are taken by the `Kitchen` class and supposed to support backwards compatibility but are not actually used/set anywhere in the function:
https://github.com/robocasa/robocasa/blob/087dfce9bb71c587d4b0ba68614a3017a66d9f72/robocasa/environments/kitchen/kitchen.py#L206-L208

snasiriany, 2024-08-26:
Hi @lorandcheng, yes this was done by design. We wanted to inherit all of the constructor args from robosuite, but there are a few that are used in robosuite but not in RoboCasa. We are opting to keep these for the time being but may drop them in the future.

closed 2024-08-26 by snasiriany

lorandcheng, 2024-08-26:
Thanks, so you don't plan to support compatibility with the GymWrapper from robosuite? Right now it throws an error because of the missing attributes. I just added `self.reward_scale=reward_scale`, etc. to avoid the error so I thought others might run into the issue as well.

snasiriany, 2024-08-28:
I see! We will add them back in and push the code today. Thanks for letting us know!