robocasa/robocasa #55: Enabling photorealistic render

sriramsk1999, 2024-08-24:
I'm trying to render photorealistic images using the `nvisii` renderer following the instructions from robosuite [documentation](https://robosuite.ai/docs/modules/renderers.html) and [demo script](https://github.com/ARISE-Initiative/robosuite/blob/master/robosuite/demos/demo_renderers.py). However, I get the following error:

```
Traceback (most recent call last):
  File "/home/sriram.sk/desktop/robocasa/teleop_exp.py", line 306, in <module>
    ep_directory, discard_traj = collect_human_trajectory_custom(
  File "/home/sriram.sk/desktop/robocasa/teleop_exp.py", line 54, in collect_human_trajectory_custom
    env.render()
  File "/home/sriram.sk/desktop/robosuite/robosuite/environments/base.py", line 519, in render
    self.viewer.render()
AttributeError: 'NoneType' object has no attribute 'render'
CUDA call (cudaFree((void*)d_pointer)) failed with code 4 (line 94): driver shutting down
terminate called after throwing an instance of 'std::runtime_error'
  what():  fatal cuda error
[1]    18430 IOT instruction (core dumped)  python teleop_exp.py
```

The initialization code:
```
    config = {
        "env_name": args.task,
        "robots": "PandaMobile",
        "controller_configs": load_controller_config(default_controller="OSC_POSE"),
        "layout_ids": args.layout,
        "style_ids": args.style,
        "translucent_robot": False,
    }

    args.renderer = "nvisii"
    
    print(colored(f"Initializing environment...", "yellow"))
    env = robosuite.make(
        **config,
        has_renderer=False,
        has_offscreen_renderer=True,
        render_camera=None,
        ignore_done=True,
        use_camera_obs=True,
        camera_depths=True,
        control_freq=20,
        renderer=args.renderer,
        camera_names=["robot0_agentview_left", "robot0_agentview_right", "robot0_eye_in_hand"]
    )
```

Am I invoking the `nvisii` renderer the wrong way or is it not supported for the `robocasa_v0.1` branch of `robosuite`?

abhihjoshi, 2024-08-24:
Could you add the code you are using here? Are you setting has_renderer to be true when creating the env? AFAIK, the nvisii renderer hasn't been updated in a while and may or may not be compatible with some of the code in robocasa. Also, I believe nvisii isn't currently actively maintained either. Maybe it could be that too (though not sure). Also, we intend to publish scripts for rendering with Omniverse which I believe includes path tracing as an option. This should be out very soon.

sriramsk1999, 2024-08-24:
Thanks for your response. I am using a modified version of the `demo_teleop.py` script, but the error can be reproduced with the default script. Yes, `has_renderer` has been set to True.

Changes to reproduce the error:
- Change renderer [here](https://github.com/robocasa/robocasa/blob/main/robocasa/demos/demo_teleop.py#L111) to `"nvisii"`
- Set `has_renderer=True` [here](https://github.com/robocasa/robocasa/blob/main/robocasa/demos/demo_teleop.py#L116)

Called with `python -m robocasa.demos.demo_teleop`

I'm running Arch Linux so my drivers are up-to-date. I did try with another Ubuntu machine as well and the issue persisted.

snasiriany, 2024-08-26:
Hi @sriramsk1999 robosuite support for the nvisii renderer has been deprecated for a while now and we are no longer supporting it. We are working on expanding our support for Omniverse, but in the meanwhile you can try running `robocasa/scripts/generate_usd_trajectories.py` on existing datasets and then rendering the generated USD files in Omniverse.

closed 2024-08-26 by snasiriany