robocasa/robocasa #56: Camera Pose

GuanxingLu, 2024-08-26:
Hi, thanks for your great work! I wonder how I can obtain the extrinsic and intrinsic matrices of a specific renderer image in robocasa?

GuanxingLu, 2024-08-28:
Hi, I looked into the robomimic and robosuite repositories and found these functions that can be obtained from a MuJoCo environment. Is that correct?
```
def get_camera_intrinsic_matrix(sim, camera_name, camera_height, camera_width):
    """
    Obtains camera intrinsic matrix.

    Args:
        sim (MjSim): simulator instance
        camera_name (str): name of camera
        camera_height (int): height of camera images in pixels
        camera_width (int): width of camera images in pixels
    Return:
        K (np.array): 3x3 camera matrix
    """
    cam_id = sim.model.camera_name2id(camera_name)
    fovy = sim.model.cam_fovy[cam_id]
    f = 0.5 * camera_height / np.tan(fovy * np.pi / 360)
    K = np.array([[f, 0, camera_width / 2], [0, f, camera_height / 2], [0, 0, 1]])
    return K

def get_camera_extrinsic_matrix(sim, camera_name):
    """
    Returns a 4x4 homogenous matrix corresponding to the camera pose in the
    world frame. MuJoCo has a weird convention for how it sets up the
    camera body axis, so we also apply a correction so that the x and y
    axis are along the camera view and the z axis points along the
    viewpoint.
    Normal camera convention: https://docs.opencv.org/2.4/modules/calib3d/doc/camera_calibration_and_3d_reconstruction.html

    Args:
        sim (MjSim): simulator instance
        camera_name (str): name of camera
    Return:
        R (np.array): 4x4 camera extrinsic matrix
    """
    cam_id = sim.model.camera_name2id(camera_name)
    camera_pos = sim.data.cam_xpos[cam_id]
    camera_rot = sim.data.cam_xmat[cam_id].reshape(3, 3)
    R = make_pose(camera_pos, camera_rot)

    # IMPORTANT! This is a correction so that the camera axis is set up along the viewpoint correctly.
    camera_axis_correction = np.array(
        [[1.0, 0.0, 0.0, 0.0], [0.0, -1.0, 0.0, 0.0], [0.0, 0.0, -1.0, 0.0], [0.0, 0.0, 0.0, 1.0]]
    )
    R = R @ camera_axis_correction
    return R

def get_z_near_far(sim):
    extent = sim.model.stat.extent  # 15
    far = sim.model.vis.map.zfar * extent
    near = sim.model.vis.map.znear * extent
    return near, far
```

abhihjoshi, 2024-08-28:
Yes, I think this should be right. I also believe Robomimic has a [function](https://github.com/ARISE-Initiative/robomimic/blob/4690f3ecfa658e1848b2e3fcf8dc2db350f9ffb9/robomimic/scripts/dataset_states_to_obs.py#L167C5-L167C20) that returns both the extrinsic and intrinsic matrices in a dictionary.

GuanxingLu, 2024-08-28:
Thanks for reminding me! I haven't checked the main branch of robomimic but spent a lot of time diving into the robocasa branch, where this function does not exist...

closed 2024-08-28 by GuanxingLu