robocasa/robocasa #57: Easiest way to avoid rendering objects

karthikm-0, 2024-08-26:
I'm trying to take screenshots of my environment without rendering the foreground objects that are part of cfgs. Would you suggest any methods that allow me to achieve this? I've tried scaling the object to near-zero using set_scale but that does not seem to do anything.

abhihjoshi, 2024-08-26:
You could maybe change the position of the objects you do not want to render to be somewhere offscreen

karthikm-0, 2024-08-26:
Thanks @abhihjoshi! Maybe I'm missing something, but setting this in the environment within the env update loop doesn't lead to any changes (set_pos or set_scale).

abhihjoshi, 2024-08-26:
Maybe something like [this](https://github.com/ARISE-Initiative/robosuite/blob/a195e8e95e9503e444f16d93b44ee13c1d3917df/robosuite/environments/base.py#L477)? You could probably just call this function and pass in the object names.

karthikm-0, 2024-08-26:
It also doesn't seem to work. For context, I'm trying to render this to an on-screen window, so perhaps there is a step I am missing? I added `self.env.render()` to force a re-render but no success. I also attempted to save an observation using the off-screen camera and still see the objects.

karthikm-0, 2024-08-26:
Sure! But it looks like clear_objects won't work since the foreground objects don't appear in the mujoco objects list inside that function.

Here's my env file (copied from prewash_food_assembly):

```python
from robocasa.environments.kitchen.kitchen import *

class WashFood(Kitchen):
    def __init__(self, cab_id=FixtureType.CABINET_TOP, *args, **kwargs):
        self.cab_id = cab_id
        super().__init__(*args, **kwargs)

    def _setup_kitchen_references(self):
        super()._setup_kitchen_references()
        self.top_cabinet = self.register_fixture_ref(
            "top_cabinet", dict(id=self.cab_id)
        )
        self.sink = self.register_fixture_ref("sink", dict(id=FixtureType.SINK))
        self.counter_cab = self.register_fixture_ref(
            "counter_cab", dict(id=FixtureType.COUNTER, ref=self.top_cabinet)
        )
        self.counter_sink = self.register_fixture_ref(
            "counter_sink", dict(id=FixtureType.COUNTER, ref=self.sink)
        )
        self.stove = self.register_fixture_ref(
            "stove", dict(id=FixtureType.STOVE)
        )

        self.init_robot_base_pos = self.top_cabinet

    def get_ep_meta(self):
        ep_meta = super().get_ep_meta()
        food_name = self.get_obj_lang("food")
        ep_meta["lang"] = (
            f"Pick the {food_name} from the cabinet and place it in the bowl. Then pick the bowl and place it in the sink. Then turn on the sink facuet."
        )

        return ep_meta

    def _reset_internal(self):
        """
        Resets simulation internal configurations.
        """
        super()._reset_internal()
        self.top_cabinet.set_door_state(
            min=0.90, max=1.0, env=self, rng=self.rng
        )

    def _get_obj_cfgs(self):
        cfgs = []

        cfgs.append(
            dict(
                name="food",
                obj_groups=["vegetable", "fruit"],
                graspable=True,
                placement=dict(
                    fixture=self.top_cabinet,
                    size=(0.50, 0.20),
                    pos=(0, -1.0),
                ),
            )
        )

        cfgs.append(
            dict(
                name="bowl",
                obj_groups="bowl",
                graspable=True,
                placement=dict(
[... 453 lines cut ...]

    # Grab reference to controller config and convert it to json-encoded string
    env_info = json.dumps(config)

    t_now = time.time()
    time_str = datetime.datetime.fromtimestamp(t_now).strftime(
        "%Y-%m-%d-%H-%M-%S"
    )

    # Create the class instance
    env_instance = TestEnv(
        env=env,
        arm=args.arm,
        config=args.config,
        renderer=args.renderer,
        max_fr=args.max_fr,
        initial_state=initial_state,
        policy_socket_address=args.policy_socket_address,
        stream_camera=args.stream_camera,
        public_folder=args.public_folder,
    )

    # Run the simulation
    env_instance.run_sim()
```

karthikm-0, 2024-08-27:
@abhihjoshi I spent a bit more time on this and noticed that objects spawned through the configs inside a particular env (e.g., WashFood) are missing the attributes that set_pos ("pos") and set_scale require in their xml definitions, unlike the other objects like fixtures.

snasiriany, 2024-08-28:
Here's something much simpler you can do: set `rgba = [0, 0, 0, 0]` for the objects you want to hide: https://github.com/robocasa/robocasa/blob/087dfce9bb71c587d4b0ba68614a3017a66d9f72/robocasa/models/objects/objects.py#L48

closed 2024-08-28 by snasiriany

karthikm-0, 2024-08-28:
@snasiriany Thanks for suggesting this! It works during initialization but I'd like to change the rgba during the simulation as well, but this approach doesn't work for that.

karthikm-0, 2024-08-28:
EDIT: It looks like the easiest way is to modify the object_placements inside the env like so:

```python
            # Loop through all objects and reset their positions
            for obj_pos, obj_quat, obj in self.env.object_placements.values():
                self.env.sim.data.set_joint_qpos(
                    obj.joints[0],
                    np.concatenate(
                        [np.array([99, 99, 99]), np.array(obj_quat)]
                    ),
                )
```

snasiriany, 2024-08-28:
Nice!