robocasa/robocasa #60: Problems with scene export

Gary3410, 2024-08-28:
Thank you for your excellent work.

I want to capture some demos in Robocasa, such as Figure 3 in the paper to demonstrate the overall layout of the scene.
Can the scenes be exported as OBJ or USD files? Or what is the collection method for Figure 3?

abhihjoshi, 2024-08-28:
Yes, all scenes in the figures were exported to USD. You can find the script to do so here: robocasa/scripts/generate_usd_trajectories.py. We are also currently writing a script to automatically save images of the scenes.

snasiriany, 2024-08-28:
Feel free to re-open if you have followup questions.

Gary3410, 2024-08-28:
Thank you for your reply.

When I tried the provided script, I found that this file is missing.
https://github.com/robocasa/robocasa/blob/087dfce9bb71c587d4b0ba68614a3017a66d9f72/robocasa/scripts/generate_usd_trajectories.py#L104

It doesn't seem to be in the `Robocasa` branch of the `Robomimic` repository.

snasiriany, 2024-08-29:
@abhihjoshi how can we fix the issue mentioned here?

abhihjoshi, 2024-08-29:
Ahh yes, you can remove that argument altogether. That was just something small I was originally testing that should've been removed. Although I believe the script should work fine even with that argument there (I don't think it is being used anywhere).

That being said, we have a new script coming out very soon that should replace this one altogether and is updated for the latest version of RoboCasa.

closed 2024-08-29 by snasiriany