robocasa/robocasa #61: an issue aboutnew layouts

hahahalove57, 2024-08-31:
Thank you for your outstanding contribution! Since I want to add new layouts to robocasa, I'm curious how all the code for layout.yaml is generated in robocasa? For example, if I get the json code for a scene by using a certain scene generation algorithm, how do I convert it to yaml so that it can be read by robocasa's demo_scenes.py in the correct format to be utilized in the same way as kitchen? I.e. how do I convert the json file to a yaml file that can be legally read by robocasa?

hahahalove57, 2024-08-31:
[bedrooms_2.json](https://github.com/user-attachments/files/16824886/bedrooms_2.json)
heres the json file

snasiriany, 2024-09-03:
Are you just looking to convert a json file to yaml file? There are some suggested solutions here: https://stackoverflow.com/questions/15941996/dump-json-into-yaml

Abc12128, 2024-10-06:
> 感谢您的杰出贡献！因为我想向 robocasa 添加新的布局，所以我很好奇 robocasa 中 layout.yaml 的所有代码是如何生成的？例如，如果我使用某个场景生成算法获得了场景的 json 代码，我该如何将其转换为 yaml，以便 robocasa 的 demo_scenes.py 能够以正确的格式读取它，以便以与厨房相同的方式使用？即，我如何将 json 文件转换为 robocasa 可以合法读取的 yaml 文件？
Hi, can you read JSON correctly now? Or can you read it correctly after converting JSON to YAML format?

Abc12128, 2024-10-06:
> Thank you for your outstanding contribution! Since I want to add new layouts to robocasa, I'm curious how all the code for layout.yaml is generated in robocasa? For example, if I get the json code [...]

Hi, can you read JSON correctly now? Or can you read it correctly after converting JSON to YAML format?