robocasa/robocasa #63: Loading environments with many objects

karthikm-0, 2024-09-04:
I was wondering if there's a way to speed up loading of environments within an on-screen renderer. I'm noticing significant delays when the environment contains more than 3 objects. Is this because the gpu rendering is only possible off-screen?

snasiriany, 2024-09-05:
I ran some benchmarking on my M1 mac:

KitchenDemo task, 1 object, with rendering:
reset time: 4.09s
fps: 29.99

KitchenDemo task, 8 objects, with rendering:
reset time: 5.17s
fps: 23.74

KitchenDemo task, 1 object, without rendering:
reset time: 3.39s
fps: 77.68

KitchenDemo task, 8 objects, without rendering:
reset time: 4.35s
fps: 57.14

The speed does drop, roughly 25% (with rendering) when going from 1 to 8 objects. This can be optimized further but must be done on the MuJoCo side. Hopefully they will address this in future releases.

Rendering is also a huge overhead, at least according to this test. We will look to see how we can reduce the overhead here.

closed 2024-09-05 by snasiriany

karthikm-0, 2024-09-05:
Thanks for benchmarking this @snasiriany! The reset times are excellent, but do you have a sense of how long the initial rendering of the environment takes? It could be a problem on my end because it takes me minutes with several objects.

snasiriany, 2024-09-06:
The initial rendering I believe is accounted for in the reset time. Can you describe your use case further? Do you have just a few object or dozens?

With 3 or fewer objects are you observing similar speeds to what I posted?

I don't think adding more objects should really impact rendering speed, it should affect the physics speed much more.

karthikm-0, 2024-09-10:
My environments contain around 10-15 objects. However, I noticed that some of these take longer to load than others; for instance the inclusion of bowls significantly slows down rendering even when there are fewer bowls than other items (e.g., 10 fruits vs 2 bowls  + 3 fruits). However, even with 3 or fewer objects it's over the times you've quoted.

snasiriany, 2024-09-11:
Can you do a test with this script I recently pushed out: https://github.com/robocasa/robocasa/blob/main/robocasa/scripts/bench_speed.py

eg. `python robocasa/scripts/bench_speeds.py --env <env-name>`

Running that does benchmarking with rendering, and running `--no_render` disables rendering. I want to see if the slowdowns you are experiencing are rendering specific or physics specific.

karthikm-0, 2024-09-12:
Thanks! Here are my numbers:

KitchenDemo task, 1 object, with rendering:
reset time: 8.76s
fps: 30.75

KitchenDemo task, 8 object, with rendering:
reset time: 14.17s
fps: 19.75

KitchenDemo task, 1 object, without rendering:
reset time: 5.48s
fps: 33.13

KitchenDemo task, 8 object, without rendering:
reset time: 9.80s
fps: 26.89

PrewashAssembly task, with rendering:
reset time: 12.32s
fps: 27.16

PrewashAssembly task, without rendering:
reset time: 9.88s
fps: 30.99

These numbers seem reasonable. i wonder if it is because I'm showing the rendering on-screen? When running the same env with collect_demos.py the load time was around 30 seconds. 

I also ran the benchmark with additional bowl objects and the time dramatically climbs.

PrewashAssembly task, rendering w/ 1 extra bowl object (single run as the next ep didn't load even after a couple of minutes)
ep #1
   26.64s reset time
   27.92 fps

After adding a 2nd bowl object, the benchmark isn't running at all (after 5 minutes of waiting).

snasiriany, 2024-09-13:
I'll investigate why the benchmark isn't running after adding objects. Might be because it can't place all the objects onthe table. But it's interesting how your reset times are so much lower than mine, and how disabling rendering doesn't make them that much faster. What computer OS and hardware are you using?

snasiriany, 2024-09-13:
Ok this makes sense. The speed will be noticeably lower if you use CPU for rendering. Can you try again with GPU rendering?

karthikm-0, 2024-09-13:
Thanks! I tried with the GPU and it was a few seconds faster:

KitchenDemo task, 1 object, with rendering:
reset time: 8.37s
fps: 31.39      

KitchenDemo task, 8 object, with rendering:
reset time: 11.57s
fps: 25.13

Is it possible to use the gpu for onscreen rendering as well?

snasiriany, 2024-09-13:
Nice, that's much better. GPU should be used for onscreen rendering as well.

karthikm-0, 2024-09-13:
> I'll investigate why the benchmark isn't running after adding objects. Might be because it can't place all the objects onthe table. But it's interesting how your reset times are so much lower than [...]

I tested your theory and I think objects are likely spawning on top of each other leading to  many collisions. Spawning them away from each other seems to work well. Thanks again for your help!