robocasa/robocasa #65: Mimicgen data download issue for the TurnOnMicrowave task

byungokhan, 2024-09-05:
When downloading mimigen data of the TurnOnMicrowave task,
an error occurred: 

raise ContentTooShortError(
urllib.error.ContentTooShortError: <urlopen error retrieval incomplete: got only 7486691982 out of 26085652712 bytes>

Is the data correct?
Thank you.

byungokhan, 2024-09-05:
I successfully downloaded the data. close this issue..

closed 2024-09-05 by byungokhan

CS-FantaC, 2025-03-17:
> I successfully downloaded the data. close this issue..

may I ask how you solve the problem,I also met problem when downloading mimicgen dataset,here is my proplem
ssl.SSLEOFError: [SSL: UNEXPECTED_EOF_WHILE_READING] EOF occurred in violation of protocol (_ssl.c:1007)
urllib.error.URLError: <urlopen error [SSL: UNEXPECTED_EOF_WHILE_READING] EOF occurred in violation of protocol (_ssl.c:1007)>

Traceback (most recent call last):
  File "/root/miniconda3/envs/robomimic_robocasa/lib/python3.10/urllib/request.py", line 1348, in do_open
    h.request(req.get_method(), req.selector, req.data, headers,
  File "/root/miniconda3/envs/robomimic_robocasa/lib/python3.10/http/client.py", line 1283, in request
    self._send_request(method, url, body, headers, encode_chunked)
  File "/root/miniconda3/envs/robomimic_robocasa/lib/python3.10/http/client.py", line 1329, in _send_request
    self.endheaders(body, encode_chunked=encode_chunked)
  File "/root/miniconda3/envs/robomimic_robocasa/lib/python3.10/http/client.py", line 1278, in endheaders
    self._send_output(message_body, encode_chunked=encode_chunked)
  File "/root/miniconda3/envs/robomimic_robocasa/lib/python3.10/http/client.py", line 1038, in _send_output
    self.send(msg)
  File "/root/miniconda3/envs/robomimic_robocasa/lib/python3.10/http/client.py", line 976, in send
    self.connect()
  File "/root/miniconda3/envs/robomimic_robocasa/lib/python3.10/http/client.py", line 1455, in connect
    self.sock = self._context.wrap_socket(self.sock,
  File "/root/miniconda3/envs/robomimic_robocasa/lib/python3.10/ssl.py", line 513, in wrap_socket
    return self.sslsocket_class._create(
  File "/root/miniconda3/envs/robomimic_robocasa/lib/python3.10/ssl.py", line 1104, in _create
    self.do_handshake()
  File "/root/miniconda3/envs/robomimic_robocasa/lib/python3.10/ssl.py", line 1375, in do_handshake
    self._sslobj.do_handshake()
ssl.SSLEOFError: [SSL: UNEXPECTED_EOF_WHILE_READING] EOF occurred in violation of protocol (_ssl.c:1007)

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/root/miniconda3/envs/robomimic_robocasa/lib/python3.10/runpy.py", line 196, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "/root/miniconda3/envs/robomimic_robocasa/lib/python3.10/runpy.py", line 86, in _run_code
    exec(code, run_globals)
  File "/root/workspace/datasets/robocasa/robocasa/scripts/download_datasets.py", line 102, in <module>
    download_datasets(
  File "/root/workspace/datasets/robocasa/robocasa/scripts/download_datasets.py", line 52, in download_datasets
    download_url(
  File "/root/workspace/datasets/robocasa/robocasa/scripts/download_kitchen_assets.py", line 115, in download_url
    urllib.request.urlretrieve(url, filename=file_to_write, reporthook=t.update_to)
  File "/root/miniconda3/envs/robomimic_robocasa/lib/python3.10/urllib/request.py", line 241, in urlretrieve
    with contextlib.closing(urlopen(url, data)) as fp:
  File "/root/miniconda3/envs/robomimic_robocasa/lib/python3.10/urllib/request.py", line 216, in urlopen
    return opener.open(url, data, timeout)
  File "/root/miniconda3/envs/robomimic_robocasa/lib/python3.10/urllib/request.py", line 519, in open
    response = self._open(req, data)
  File "/root/miniconda3/envs/robomimic_robocasa/lib/python3.10/urllib/request.py", line 536, in _open
    result = self._call_chain(self.handle_open, protocol, protocol +
  File "/root/miniconda3/envs/robomimic_robocasa/lib/python3.10/urllib/request.py", line 496, in _call_chain
    result = func(*args)
  File "/root/miniconda3/envs/robomimic_robocasa/lib/python3.10/urllib/request.py", line 1391, in https_open
    return self.do_open(http.client.HTTPSConnection, req,
  File "/root/miniconda3/envs/robomimic_robocasa/lib/python3.10/urllib/request.py", line 1351, in do_open
    raise URLError(err)
urllib.error.URLError: <urlopen error [SSL: UNEXPECTED_EOF_WHILE_READING] EOF occurred in violation of protocol (_ssl.c:1007)>