robocasa/robocasa #67: Questions about Robocasa: ROS Integration, Hello Robot Stretch2 Support, and Camera-only Simulation

ilileun, 2024-09-09:
Hello,
I have a few questions regarding Robocasa's current features and future plans:

- ROS Integration: Are there any plans to integrate ROS1 or ROS2 with Robocasa? If not officially planned, are there any current methods or workarounds to use Robocasa with ROS?
- Hello Robot Stretch2 Support: Is there any intention to add support for the Hello Robot Stretch2 in Robocasa? This could expand the platform's versatility for various robotics applications.
- Camera-only Simulation: Is it possible to create a simulation where the robot itself is invisible, but we can still move a camera around the environment as if it were mounted on the robot? If so, could you provide some guidance on how to achieve this?

Any information or insights on these topics would be greatly appreciated. Thank you for your time and consideration!

snasiriany, 2024-09-12:
Hi @ilileun,

ROS integration is typically used for real world robotics applications, to connect with sensors, etc. Can you describe your use case further?

Hello Stretch 2 robot will likely be supported but not in the immediate next few weeks. At the moment we are focusing on improving the capabilities with our existing robots.

Yes, I think there's a way to "hide" the robot. I think using the `translucent_robot` option and setting the value in this line to `0.0` should do it: https://github.com/robocasa/robocasa-dev/blob/bfac4c1bad1a06bef4c5c30ba2e96ecd4f8e16ad/robocasa/environments/kitchen/kitchen.py#L1261

closed 2024-09-12 by snasiriany

ilileun, 2024-09-17:
Thank you. When I followed your previous instructions, it seems the entire robot disappeared.
However, what I want is to spawn only the camera. How can I make the gripper itself invisible as well?

[image]

snasiriany, 2024-09-18:
Ah. To hide the gripper as well you can change the lines in that function to:
```
        super().visualize(vis_settings=vis_settings)

        visual_geom_names = []

        for robot in self.robots:
            robot_model = robot.robot_model
            visual_geom_names += robot_model.visual_geoms

            gripper_model = robot.gripper
            for arm in gripper_model:
                visual_geom_names += gripper_model[arm].visual_geoms

        for name in visual_geom_names:
            rgba = self.sim.model.geom_rgba[self.sim.model.geom_name2id(name)]
            if self.translucent_robot:
                rgba[-1] = 0.0
            else:
                rgba[-1] = 1.0
 ```