robocasa/robocasa #68: Customizing Object

ziyi105, 2024-09-12:
Thanks for your amazing work! 

I am trying to customize a chair object, but I am not sure how to do it. Could you please give me some tips on how to do it? Specifically, how does the code load objects from Objaverse and how can I specify a path to a chair object from Objaverse in the code. 

I tried to back-trace the demo_objects function but unfortunately it could not display properly on my laptop, below is a screenshot of what I see when I run the command. This issue persists even if I press N to view another object.

[image]

snasiriany, 2024-09-12:
Hi @ziyi105,

Are you trying to import a new model from Objaverse into RoboCasa? Or is this an existing chair that we provided already?

For importing new objects, there's a process which involves processing the raw files into submeshes, generating bounding boxes, etc. We based the code for this off of and adapted it for our purposes: https://github.com/kevinzakka/obj2mjcf. Our object import code is not public at this moment.

We do provide some references to existing stools under `robocasa/models/assets/fixtures/accessories/stools` which you can access after you run the initial script to download all the assets (`python robocasa/scripts/download_kitchen_assets.py`)

closed 2024-09-12 by snasiriany

ziyi105, 2024-09-14:
Thanks for your reply! 

Is there a way to fix the problem where demo object cannot be shown properly like the in screenshot I provided?

snasiriany, 2024-09-17:
Can you share additional details on what script you ran exactly, and the computing platform, OS version, mujoco version, whether GPU is used, etc?

quas-modo, 2024-10-11:
> Hi @ziyi105,
>
> Are you trying to import a new model from Objaverse into RoboCasa? Or is this an existing chair that we provided already?
>
> For importing new objects, there's a process which involves [...]

Hi,

I’m in the process of customizing new objects, and I noticed that you mentioned using **obj2mjcf** for this purpose. Could you kindly share which specific parameters or options you’ve used with obj2mjcf during the process? Any insights would be greatly appreciated.

ziyi105, 2024-10-14:
hi @quas-modo 
I did not end up using obj2mjcf in the end as the object i needed is available in fixture accessories (stool), hence i edited the xml file of stool fixture and add it to kitchen_objects.py

ziyi105, 2024-10-14:
> Can you share additional details on what script you ran exactly, and the computing platform, OS version, mujoco version, whether GPU is used, etc?

@snasiriany  Sorry I missed your reply. I ran demo_objects.py on windows 11 and no GPU is used. Mujoco verison is 3.1.1

snasiriany, 2024-10-14:
@ziyi105 do the other demos scripts work for you?

See here: https://robocasa.ai/docs/introduction/quick_start.html

ziyi105, 2024-10-17:
> @ziyi105 do the other demos scripts work for you?
> 
> See here: https://robocasa.ai/docs/introduction/quick_start.html

Yeah, other demo scripts worked for me, only demo_objects.py didnt work

snasiriany, 2024-10-17:
Another manual option to try, run `python -m mujoco.viewer --mjcf <path-to-object-xml-file>`

Eg. the xml path can be `robocasa/models/assets/objects/objaverse/boxed_drink/boxed_drink_2/model.xml`

ThisisXXZ, 2025-07-05:
> [@snasiriany](https://github.com/snasiriany) Hi, I am trying to use obj2mjcf and I noticed robocasa follows robosuite robosuite's xml structure to import objects. The question is how I can get the bottom_site, top_site and horizontal_radius_site for custom objects. Many thanks!

Same problem here. May I ask if you've came up with a solution on that? Really have no idea how to get these three attributes