robocasa/robocasa #70: How to control the robot to move around the map

ziyi105, 2024-09-14:
Hi,

I am working on creating new datasets where the robot moves to a specific point, grabs an object, and then moves to another point. However, I'm unsure how to control the robot's movement around the map since the available control options seem to only apply to the robot's arm.

Could you advise on how to control the robot's overall movement, including navigation?

snasiriany, 2024-09-15:
For the default PandaMobile robot paired with OSC controller, here’s how the 12 dim action space works (zero indexed):

dim 0/1/2/3/4/5: arm translation and rotation
dim 6: gripper
dim 7/8/9/10: base
dim 11: mode controlling whether to use arm or base. if you set it to -1 it should control the arm for sure. Set it to 1 to control base instead

So in your case, you'd want to set dim 11 to 1.0, and play with dims 7/8/9/10.

closed 2024-09-15 by snasiriany

ziyi105, 2024-09-15:
Thank you so much for your reply!