robocasa/robocasa #71: reset_to() function creates new objects. Deterministic testing

SoshinK, 2024-09-15:
Hello! Thank you for excellent work! However, I can't figure out how to recreate the environment with the same objects in pick-and-place tasks. 

For instance, I manually create a new environment and save its state using get_state() function:

```
env_meta = {'env_name': 'PnPCabToCounter',
 'env_version': '1.4.1',
 'type': 1,
 'env_kwargs': {'robots': 'PandaMobile',
  'controller_configs': {'type': 'OSC_POSE',
   'input_max': 1,
   'input_min': -1,
   'output_max': [0.05, 0.05, 0.05, 0.5, 0.5, 0.5],
   'output_min': [-0.05, -0.05, -0.05, -0.5, -0.5, -0.5],
   'kp': 150,
   'damping_ratio': 1,
   'impedance_mode': 'fixed',
   'kp_limits': [0, 300],
   'damping_ratio_limits': [0, 10],
   'position_limits': None,
   'orientation_limits': None,
   'uncouple_pos_ori': True,
   'control_delta': True,
   'interpolation': None,
   'ramp_ratio': 0.2},
  'camera_names': ['robot0_agentview_left',
   'robot0_agentview_right',
   'robot0_eye_in_hand'],
  'camera_heights': 128,
  'camera_widths': 128,
  'has_renderer': False,
  'has_offscreen_renderer': True,
  'env_name': 'PnPCabToCounter',
  'seed': 43},
 'env_lang': None}

env_kwargs = dict(
        env_meta=env_meta,
        env_name='PnPCabToCounter',
        render=False,
        render_offscreen=True,
        use_image_obs=True,
        seed=42,
    )

env = EnvUtils.create_env_from_metadata(**env_kwargs)

_ = env.reset()
state_dict = env.get_state()
```

However, when I load environment from this state, I got new objects in `object_cfg`, and accordingly wrong language instruction:

```
>>> _ = env.reset()
>>> _ = env.reset_to(state_dict)
>>> env._ep_lang_str
'pick the hot dog from the cabinet and place it on the counter'
```

But original instruction in state_dict is:
```
>>> json.loads(state_dict['ep_meta'])['lang']
'pick the cereal from the cabinet and place it on the counter'
```

And, rendered camera view clearly shows that scene does not contain new generated object (no hot dog):

[image]

Is this expected behavior? And if so, how can I recreate the same scene every time for deterministic testing?

Thank you

snasiriany, 2024-09-18:
Thanks for raising this! The issue was that while we load the xml from state_dict correctly, and most meta data information from the state_dict (layout id, object categories), we were not loading the exact identity of the objects used - we were resampling them.

These are the new changes I added to fix this: https://github.com/robocasa/robocasa/blob/daf2fb87ce49c3c83e1d9941b51af5e3c7fe5d42/robocasa/environments/kitchen/kitchen.py#L432-L443

This should be fixed now. Pull the latest from the robocasa codebase and all the robosuite branch associated with it.

SoshinK, 2024-09-18:
Cool! Thanks!

SoshinK, 2024-09-20:
It seems that also left/right drawer's placement is broken in CloseDrawer task.

```
env_kwargs = dict(
        env_meta=env_meta,
        env_name='CloseDrawer',
        render=False,
        render_offscreen=True,
        use_image_obs=True,
        seed=54,
    )

env = EnvUtils.create_env_from_metadata(**env_kwargs)

_ = env.reset()
state_dict = env.get_state()
```

After resetting:
```
>>> _ = env.reset()
>>> _ = env.reset_to(state_dict)
>>> env._ep_lang_str
'close the right drawer'
>>> json.loads(state_dict['ep_meta'])['lang']
'close the left drawer'
```