robocasa/robocasa #72: Questions about deploying models to physical robots

mrvgao, 2024-09-16:
Hello, Thanks for all your work and building for this community!

Recently, I completed some new models in Robocasa,and I want to deploy these models to real physical robots. But I didn't find any document or tutorial for this deployment. 

Would you please share some information about how to deploy it? 

Thanks!

snasiriany, 2024-09-16:
We used the [DROID hardware setup](https://droid-dataset.github.io/) to do our real world experiments.

They have a policy learning repo that's based on robomimic where they provide evaluation scripts: https://github.com/droid-dataset/droid_policy_learning/tree/master?tab=readme-ov-file#evaluating-trained-policies

Their evaluation scripts are a useful reference for deploying robomimic policies to their physical hardware.

mrvgao, 2024-09-17:
thank you so much! I will try it.