robocasa/robocasa #74: Green and Red glitched textures

eric-li-2, 2024-09-22:
I followed the install instructions for robocasa on a fresh headless ubuntu instance on ec2.
I also installed the following packages: 
- sudo apt install libglew-dev 
- pip install imageio[ffmpeg]

This resulted in the following output when rendering videos

[image]

I suspect that it is possibly due to the bvactive tag from mujoco, but I am not sure. For info on the bvactive tag, see:
https://mujoco.readthedocs.io/en/3.1.2/changelog.html

Could someone knowledgeable please help?

snasiriany, 2024-09-22:
This is actually showing all of the collision geoms, this should normally be disabled.

Are you using MuJoCo 3.1.1? That's the stable version we've been developing on.

If you can share a full set of code for me to reproduce your video I can help further.

eric-li-2, 2024-09-23:
I fixed it, it was my fault. I installed everything according to instructions. The numpy version automatically installed was 1.23.2 which was too new (incompatible with some of the robosuite code).

I didn't know this at the time, so I resolved a compatibility issue in robosuite: https://github.com/ARISE-Initiative/robosuite/blob/09d942e01acf312ec8b3da1d6af892c876798f0d/robosuite/utils/transform_utils.py#L414
by changing the line to `M = np.asarray(rmat, dtype=np.float32)[:3, :3]`

I am not sure if my one line fix broke the code, but the most important thing was to downgrade numpy down to 1.23.1

snasiriany, 2024-09-23:
The current version of numpy I have installed is numpy==1.23.2, which has been working for me. Can you elaborate further where you think numpy==1.23.2 is causing issues for you? That line in transform_utils.py you mentioned - it doesn't seem to be causing issues on my end.

eric-li-2, 2024-09-23:
I think it is causing numpy issues because I was getting errors in robocasa, not just robomimic and robosuite.

For example, in the train.py script, attempting to instantiate a dataloader from a dataset resulted in an error. This was due to the default pytorch data collator attempting to cast the raw numpy ndarrays into tensors with torch.as_tensor(...) before concatenating. 

The error I got back was "expected np.ndarray but got numpy.ndarray instead" when calling as_tensor. 
Here is a relevant pytorch discussion thread:
https://discuss.pytorch.org/t/why-do-i-get-typeerror-expected-np-ndarray-got-numpy-ndarray-when-i-use-torch-from-numpy-function/37525

In it, users stated that downgrading numpy seemed to work. So this is what I followed.

snasiriany, 2024-09-23:
Ok thanks for letting me know. We are migrating everything to numpy 2.0, and I will make sure robomimic works with numpy 2.0.

Closing issue for now.

closed 2024-09-23 by snasiriany

xuanshao27, 2024-10-09:
I also find the same issue when I try to train BC-Transformer policy follow the doc. the traceback said that 
"robomimic/robomimic/models/base_nets.py", line 1380, in __init__
    pos_x = torch.from_numpy(pos_x.reshape(1, self._in_h * self._in_w)).float()
TypeError: expected np.ndarray (got numpy.ndarray)
I follow the advice as eric said to "Downgrading to 1.23.1 fixed everything",then I solved the issue.

LeiHanyue, 2024-10-23:
After downgrading NumPy to version 1.23.1, an error emerged in the eval_utils.py file. I did not make any changes to the code. My system is running Ubuntu 22.04. The error message is provided below. The confusing thing is that even after I reinstalled NumPy version 1.23.2, the issue still persists, leaving me puzzled.
Traceback (most recent call last):
  File "/home/miniconda3/envs/robocasa/lib/python3.10/runpy.py", line 196, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "/home/miniconda3/envs/robocasa/lib/python3.10/runpy.py", line 86, in _run_code
    exec(code, run_globals)
  File "/home/.vscode/extensions/ms-python.debugpy-2024.8.0-linux-x64/bundled/libs/debugpy/adapter/../../debugpy/launcher/../../debugpy/__main__.py", line 39, in <module>
    cli.main()
  File "/home/.vscode/extensions/ms-python.debugpy-2024.8.0-linux-x64/bundled/libs/debugpy/adapter/../../debugpy/launcher/../../debugpy/../debugpy/server/cli.py", line 430, in main
    run()
  File "/home/.vscode/extensions/ms-python.debugpy-2024.8.0-linux-x64/bundled/libs/debugpy/adapter/../../debugpy/launcher/../../debugpy/../debugpy/server/cli.py", line 284, in run_file
    runpy.run_path(target, run_name="__main__")
  File "/home/.vscode/extensions/ms-python.debugpy-2024.8.0-linux-x64/bundled/libs/debugpy/_vendored/pydevd/_pydevd_bundle/pydevd_runpy.py", line 321, in run_path
    return _run_module_code(code, init_globals, run_name,
  File "/home/.vscode/extensions/ms-python.debugpy-2024.8.0-linux-x64/bundled/libs/debugpy/_vendored/pydevd/_pydevd_bundle/pydevd_runpy.py", line 135, in _run_module_code
    _run_code(code, mod_globals, init_globals,
  File "/home/.vscode/extensions/ms-python.debugpy-2024.8.0-linux-x64/bundled/libs/debugpy/_vendored/pydevd/_pydevd_bundle/pydevd_runpy.py", line 124, in _run_code
    exec(code, run_globals)
  File "/home/code_bot/robocasa/robocasa/utils/zmq_eval_utils.py", line 102, in <module>
    env = create_eval_env(env_name=env_name, )
  File "/home/code_bot/robocasa/robocasa/utils/zmq_eval_utils.py", line 58, in create_eval_env
    env = robosuite.make(**env_kwargs)
  File "/home/code_bot/robosuite/robosuite/environments/base.py", line 41, in make
    return REGISTERED_ENVS[env_name](*args, **kwargs)
  File "/home/code_bot/robocasa/robocasa/environments/kitchen/single_stage/kitchen_doors.py", line 161, in __init__
    super().__init__(door_id=door_id, *args, **kwargs)
  File "/home/code_bot/robocasa/robocasa/environments/kitchen/single_stage/kitchen_doors.py", line 151, in __init__
    super().__init__(behavior="close", *args, **kwargs)
  File "/home/code_bot/robocasa/robocasa/environments/kitchen/single_stage/kitchen_doors.py", line 21, in __init__
    super().__init__(*args, **kwargs)
  File "/home/code_bot/robocasa/robocasa/environments/kitchen/kitchen.py", line 286, in __init__
    super().__init__(
  File "/home/code_bot/robosuite/robosuite/environments/manipulation/manipulation_env.py", line 184, in __init__
    super().__init__(
  File "/home/code_bot/robosuite/robosuite/environments/robot_env.py", line 227, in __init__
    super().__init__(
  File "/home/code_bot/robosuite/robosuite/environments/base.py", line 144, in __init__
    self._load_model()
  File "/home/code_bot/robocasa/robocasa/environments/kitchen/kitchen.py", line 372, in _load_model
    fxtr_placements = fxtr_placement_initializer.sample()
  File "/home/code_bot/robocasa/robocasa/utils/placement_samplers.py", line 557, in sample
    new_placements = sampler.sample(placed_objects=placed_objects, **s_args)
  File "/home/code_bot/robocasa/robocasa/utils/placement_samplers.py", line 329, in sample
    mat2quat(euler2mat([0, 0, self.reference_rot])), to="wxyz"
TypeError: can't unbox array from PyObject into native value.  The object maybe of a different type

LeiHanyue, 2024-10-23:
I found that there were two versions of NumPy in my environment: one was 1.23.2 and the other was 2.0.1. I uninstalled both of them and then reinstalled the 1.23.2 version. After that, the color problem and the problems described above disappeared.

snasiriany, 2024-10-28:
@LeiHanyue great to know, thanks! For others looking at this thread: these versions all generally work well for me:
```
pip install numba==0.56.4 numpy==1.23.3 mujoco==3.1.1
```