robocasa/robocasa #80: An issue about the "--render_offscreeen"

KysonYoshi, 2024-09-25:
Hi, I've encountered an issue with the "--render_offscreen". 
I can successfully run a demo task with offscreen rendering by using 
python -m robocasa.demos.demo_tasks --render_offscreen.
However, when I checked the output .mp4 files, the rendering quality didn't seem well.

Any help to fix this would be appreciated!

https://github.com/user-attachments/assets/5eff2f9a-b9c1-4bb7-9613-f115e16279bb

https://github.com/user-attachments/assets/96486b5f-924b-4dc4-870a-ceef1b827baf

snasiriany, 2024-09-26:
Interesting. What's happening here is that you are seeing both the visual and collision geoms of the model. The collision geoms should be hidden. Can you share some more context?
1. What OS are you running?
2. What version of MuJoCo?
3. Did you make any changes to the code base?

KysonYoshi, 2024-09-26:
Thank you for the response!

1. What OS are you running?
>>> I'm running robocasa in Windows WSL.
2. What version of MuJoCo?
>>> I'm using MuJoCo 3.1.1
3. Did you make any changes to the code base?
>>> Yes, I've changed the code (/robosuite/robosuite/utils/transform_utils.py):
        from
        M = np.array(rmat, dtype=np.float32, copy=False)[:3, :3]
        to 
        M = np.array(rmat, dtype=np.float32)[:3, :3]
        
        otherwise, it'll output this error:
        "M = np.array(rmat, dtype=np.float32, copy=False)[:3, :3]
        ValueError: Unable to avoid copy while creating an array as requested.
        If using `np.array(obj, copy=False)` replace it with `np.asarray(obj)` to allow a copy when needed (no behavior change in NumPy 1.x).
        For more details, see https://numpy.org/devdocs/numpy_2_0_migration_guide.html#adapting-to-changes-in-the-copy-keyword."

snasiriany, 2024-09-27:
I'm also seeing the issue on certain workspaces. I will propose a solution soon.

snasiriany, 2024-09-27:
Try downgrading to numpy 1.23.2, that should solve the issue. `pip install numpy==1.23.2`

closed 2024-09-27 by snasiriany

ziyi105, 2024-10-25:
Hi! I am also facing similar issue when i use offscreen renderer to capture the robot0_frontview images. My numpy  version is already 1.23.2, any idea why it shows the red and green stuff in the picture below?

[image]

This is my code:
`    env = robosuite.make(
        **config,
        has_renderer=True,
        has_offscreen_renderer=True,
        render_camera="robot0_frontview",
        ignore_done=True,
        use_camera_obs=True,
        control_freq=20,
        camera_depths=True,
        camera_names="robot0_frontview",
        renderer="mjviewer",
    )`

Thanks!

ziyi105, 2024-10-25:
Never mind, I solved the issue by adding         
render_visual_mesh=False 
in my configurations.