robocasa/robocasa #81: Question of success rate and observation set

JediWarriorZou, 2024-09-26:
Hello. I used BC-transformer and trained it for task PickPlaceCounterToSink. My training dataset is 500 generated demos. I followed robomimic BC-transformer since your paper also followed it. I used 50 trials for evaluation. When observation set used is ['robot0_agentview_left_image', 'robot0_agentview_right_image', 'robot0_base_pos', 'robot0_base_quat', 'robot0_base_to_eef_pos', 'robot0_base_to_eef_quat', 'robot0_eye_in_hand_image', 'robot0_gripper_qpos'], the success rate is only 2%. When observation set used is ["robot0_eef_pos", "robot0_eef_quat", "robot0_gripper_qpos", "object"], the success rate can reach 12% which is close to the result of this task trained on Generated-300. I wonder whether the latter observation set was used for your paper.

snasiriany, 2024-09-26:
Our Generated-300 result is training a multi-task policy over 24/25 atomic tasks, with 300 demos per task. So in essence it is 7.2k demos over all tasks. In appears that your experiment is a single task policy on PickPlaceCounterToSink with 500 demos total. I would consider running the multi-task policy over these 24 tasks and seeing if there are improvements.

closed 2024-09-26 by snasiriany

JediWarriorZou, 2024-09-26:
OK. So do you use observation set containing 3 images from cameras for training multi-task policy? Thanks.

snasiriany, 2024-09-26:
Yes, we use the 3 images. We don't use `object` for training our policies.

JediWarriorZou, 2024-09-26:
It seems that it costs quite a long time to train on 7.2k demos, since it costs a few days to train on 500 demos before.

snasiriany, 2024-09-26:
For us it takes about ~24 hours to train the policy for 1000 epochs. What kind of GPU are you using to train?

On another note: make sure that you either turn off evaluations completely or limit the set of tasks being evaluated. Otherwise it will try to evaluate all of the tasks every 100 epochs.

kailucs, 2024-11-11:
Hi, just want to follow this question, is there any pre-trained checkpoint available for our direct use? e.g. the BC-transformer where the paper mentioned, for the atomic tasks?

snasiriany, 2024-11-11:
We don't have a checkpoint that's compatible with the public repo but you can follow the instructions here to reproduce the policy learning results: https://robocasa.ai/docs/use_cases/policy_learning.html