robocasa/robocasa #82: Mesh Rendering Issues and Object Pose Information

ilileun, 2024-09-26:
## Issue 1: Mesh Rendering Problem with Large Numbers of Objects

When spawning objects randomly at desired locations (10-30 objects), everything works fine. However, when spawning 40 or more objects, the meshes don't render correctly. For example, a lemon that should be yellow appears white, indicating that its mesh isn't being applied properly.

### Current Behavior
- Spawning 10-30 objects works correctly
- Spawning 40+ objects results in incorrect mesh rendering

### Expected Behavior
- All objects should render correctly regardless of the number spawned

### Question
Is there a way to resolve this issue and ensure proper mesh rendering when spawning larger numbers of objects?

## Issue 2: Obtaining Object Pose Information

When using the "obj_groups="all" option to spawn objects, I need a way to determine which objects are being generated and their positions.

### Question
Is there a method to obtain the world coordinate poses for all objects built in the LAYOUT, including both fixed objects (e.g., drawers, counters, islands) and movable objects (e.g., apples, cereals)?

- I've attempted extensive debugging but have been unable to resolve these issues
- Any insights, suggestions, or solutions would be greatly appreciated

## Code Sample
```python
    def _get_obj_cfgs(self):
        """
        Get the object configurations for the counter to cabinet pick and place task.
        Puts the target object in the front area of the counter. Puts a distractor object on the counter
        and the back area of the cabinet.

        """
        cfgs = []
        cfgs.append(
            dict(
                name="obj",
                obj_groups=self.obj_groups,
                exclude_obj_groups=self.exclude_obj_groups,
                graspable=True,
                placement=dict(
                    fixture=self.counter,
                    sample_region_kwargs=dict(
                        ref=self.cab,
                    ),
                    size=(0.60, 0.30),
                    pos=(0.0, -1.0),
                    offset=(0.0, 0.10),
                ),
            )
        )

        # distractors
        cfgs.append(
            dict(
                name="distr_counter",
                obj_groups="all",
                placement=dict(
                    fixture=self.counter,
                    sample_region_kwargs=dict(
                        ref=self.cab,
                    ),
                    size=(1.0, 0.30),
                    pos=(0.0, 1.0),
                    offset=(0.0, -0.05),
                ),
            )
        )
        cfgs.append(
            dict(
                name="distr_cab",
                obj_groups="all",
                placement=dict(
                    fixture=self.cab,
                    size=(1.0, 0.20),
                    pos=(0.0, 1.0),
                    offset=(0.0, 0.0),
                ),
            )
        )

        # cfgs.append(
        #     dict(
        #         name="tomato",  
        #         obj_groups="tomato",  
        #         # placement=dict(
        #         #     fixture=self.counter,
        #         #     sample_region_kwargs=dict(
        #         #         ref=self.cab,
        #         #     ),
        #         #     size=(0.35, 0.2),
        #         #     pos=("ref", -1.0),
        #         # ),
        #         placement=dict(
        #             fixture=self.island,
        #             size=(0.0, 0.0),  
        #             pos=(0.0, 0.0),
        #             offset=(0.0, 0.0),
        #         ),
        #         ensure_object_boundary_in_range=True,
        #     )
        # )

        """
        Get the object configurations for the counter to cabinet pick and place task.
        Spawns 30 objects randomly on the island.
        """

        island_size = self.island.size
        spawn_object_num = 40 

        for i in range(spawn_object_num):
            cfgs.append(
                dict(
                    name=f"object_{i}",  
                    obj_groups="all",  
                    # graspable=True,
                    # cookable=True,
                    placement=dict(
                        fixture=self.island,
                        size=(
                            island_size[0] * 0.6,
                            island_size[1] * 0.6,
                        ),  
                        pos=(0.0, 0.0), 
                        offset=(0.0, 0.0),
                    ),
                    ensure_object_boundary_in_range=True,
                )
            )

        return cfgs

abhihjoshi, 2024-09-26:
> Issue 1: Mesh Rendering Problem with Large Numbers of Objects

Which version of mujoco are you using? I think versions 3.2.1 reverted the number of materials to a much smaller amount. We recommend using v3.1.1. 

> Issue 2: Obtaining Object Pose Information

You should be able to read this information from mjData. You can use this function here to do this: https://github.com/ARISE-Initiative/robosuite-dev/blob/55865c4d3c28cc392ef144d49694eee9f195e03e/robosuite/utils/binding_utils.py#L718

snasiriany, 2024-09-26:
Re 1. Mostly likely the maximum number of textures are being used. MuJoCo has a cap on this. I'm not sure we can do much about it. If you use 40+ objects the simulation will also get quite slower. It's probably good to contact the MuJoCo team about support for much larger scenes.

Re 2. Take a look here about a related question another user asked: https://github.com/robocasa/robocasa/issues/73#issuecomment-2358949818. This will give you the poses of the objects. The core functions you need are here: https://github.com/robocasa/robocasa/blob/daf2fb87ce49c3c83e1d9941b51af5e3c7fe5d42/robocasa/environments/kitchen/kitchen.py#L1157-L1164

You can follow this logic to obtain the poses of the fixtures as well, though that's not explicitly in the code right now.

closed 2024-09-26 by snasiriany

ilileun, 2024-09-26:
> > 문제 1: 다수의 객체로 인한 메시 렌더링 문제
> 
> 어떤 버전의 mujoco를 사용하고 있나요? 버전 3.2.1에서는 재료의 양이 훨씬 적은 양으로 되돌아간 것 같습니다. v3.1.1을 사용하는 것이 좋습니다.
> 
> > 문제 2: 물체 자세 정보 얻기
> 
> mjData에서 이 정보를 읽을 수 있어야 합니다. 여기에서 이 기능을 사용하여 이 작업을 수행할 수 있습니다: [ https://github.com/ARISE-Initiative/robosuite-dev/blob/55865c4d3c28cc392ef144d49694eee9f195e03e/robosuite/utils/binding_utils.py#L718 ](https://github.com/ARISE-Initiative/robosuite-dev/blob/55865c4d3c28cc392ef144d49694eee9f195e03e/robosuite/utils/binding_utils.py#L718)

I am using MuJoCo version 3.1.1... :((  
Do you know how many objects are supported in versions 3.1.1 and 3.1.2, and what the maximum number of objects that can be rendered is?"

ilileun, 2024-09-26:
[image]

> Re 1. Mostly likely the maximum number of textures are being used. MuJoCo has a cap on this. I'm not sure we can do much about it. If you use 40+ objects the simulation will also get quite slower. It's probably good to contact the MuJoCo team about support for much larger scenes.
> 
> Re 2. Take a look here about a related question another user asked: [#73 (comment)](https://github.com/robocasa/robocasa/issues/73#issuecomment-2358949818). This will give you the poses of the objects. The core functions you need are here:
> 
> https://github.com/robocasa/robocasa/blob/daf2fb87ce49c3c83e1d9941b51af5e3c7fe5d42/robocasa/environments/kitchen/kitchen.py#L1157-L1164
> 
> You can follow this logic to obtain the poses of the fixtures as well, though that's not explicitly in the code right now.

[image]

I was really impressed when I saw on your website that you had placed a huge number of objects on the island. I'd like to spawn objects in a similar way, and I'm curious about how you did it.

snasiriany, 2024-09-26:
@abhihjoshi I think MuJoCo has a limit for the number of textures in the overall model, and after a certain number of textures are used, MuJoCo will ignore them. @abhihjoshi please test on your end and let us know if you can still render > 50 objects.

ilileun, 2024-09-26:
I upload at 

> I can test this out on my end, sure. Though I think mujoco v3.1.1 has a max number of materials/textures of 1000. AFAIK, I don't think we get close to that number in robocasa, which is why I'm a little surprised by this.
> 
> @ilileun can you share the code you are using for the environment you created?

Project repositories: 
- RoboCasa (modified version): https://github.com/ilileun/robocasa 
- RoboSuite (modified version, robocasa_v0.1 branch): https://github.com/ilileun/robosuite
To run the project, use the same method as before: 
```
python -m robocasa.demos.demo_kitchen_scenes
```
Key changes: 
  1) Added RGB-D saving functionality: Press '5' on the keyboard to save RGB-D data. 
  2) Object spawning code location: 
  `https://github.com/ilileun/robocasa/blob/ac15bb94ed36132d8f94795b161b1c0cc94dbb4c/robocasa/environments/kitchen/single_stage/kitchen_pnp.py#L202`
Please note that the code might not be fully organized as various tests are still in progress. I apologize for any inconvenience this may cause.
Thank you very much for your help with this project. Your support has been invaluable in making these modifications and tests possible.