robocasa/robocasa #84: Script Not Completing when Running `dataset_states_to_obs.py`

KysonYoshi, 2024-10-03:
Hi,

I want to generate a video that explores the kitchen scene, so I decided to use `collect_demos.py.` However, I ecountered an issue with the script `dataset_states_to_obs.py`. When processing a dataset in RoboCasa, the script does not complete as expected and seems to hang, without returning control back to the command line.
### Step to Reproduce:
1. collect.demos.py line 119 
```
discard_traj = True
```
adjust to 
```
#discard_traj = True
```
Since I need a video that goes through the kitchen scene.
2. 
```
python robocasa/scripts/collect_demos.py --env PnPCounterToCab --device keyboard
```
3. 
```
python robomimic/scripts/conversion/convert_robosuite.py --dataset <ds-path>
```
4.
```
OMP_NUM_THREADS=1 MPI_NUM_THREADS=1 MKL_NUM_THREADS=1 OPENBLAS_NUM_THREADS=1 python robomimic/scripts/dataset_states_to_obs.py --dataset <ds-path>
```
The script starts running, but it never finishes or returns to the (robocasa) prompt. The following appears:
```
  (robocasa) cl6933@LAPTOP-QEKOMNA4:~/robocasa$ OMP_NUM_THREADS=1 MPI_NUM_THREADS=1 MKL_NUM_THREADS=1 OPENBLAS_NUM_THREADS=1 python /home/cl6933/robomimic/robomimic/scripts/dataset_states_to_obs.py --dataset /home/cl6933/robocasa/robocasa/models/assets/demonstrations_private/2024-10-02-20-36-12/demotest.hdf5
  ROBOMIMIC WARNING(
      No private macro file found!
      It is recommended to use a private macro file
      To setup, run: python /home/cl6933/robomimic/robomimic/scripts/setup_macros.py
  )
  input file: /home/cl6933/robocasa/robocasa/models/assets/demonstrations_private/2024-10-02-20-36-12/demotest.hdf5
  output file: /home/cl6933/robocasa/robocasa/models/assets/demonstrations_private/2024-10-02-20-36-12/demotest_im128.hdf5
  WARNING: mimicgen environments not imported since mimicgen is not installed!

  ============= Initialized Observation Utils with Obs Spec =============

  using obs modality: low_dim with keys: []
  using obs modality: rgb with keys: ['robot0_eye_in_hand_image', 'robot0_agentview_right_image', 'robot0_agentview_left_image']
  WARNING: mimicgen environments not imported since mimicgen is not installed!

  ============= Initialized Observation Utils with Obs Spec =============

  using obs modality: low_dim with keys: []
  using obs modality: rgb with keys: ['robot0_eye_in_hand_image', 'robot0_agentview_right_image', 'robot0_agentview_left_image']
  WARNING: mimicgen environments not imported since mimicgen is not installed!

  ============= Initialized Observation Utils with Obs Spec =============

  using obs modality: low_dim with keys: []
  using obs modality: rgb with keys: ['robot0_eye_in_hand_image', 'robot0_agentview_right_image', 'robot0_agentview_left_image']
  WARNING: mimicgen environments not imported since mimicgen is not installed!

  ============= Initialized Observation Utils with Obs Spec =============

  using obs modality: low_dim with keys: []
  using obs modality: rgb with keys: ['robot0_eye_in_hand_image', 'robot0_agentview_right_image', 'robot0_agentview_left_image']
  WARNING: mimicgen environments not imported since mimicgen is not installed!

  ============= Initialized Observation Utils with Obs Spec =============

  using obs modality: low_dim with keys: []
  using obs modality: rgb with keys: ['robot0_eye_in_hand_image', 'robot0_agentview_right_image', 'robot0_agentview_left_image']
  libEGL warning: failed to open /dev/dri/renderD128: Permission denied

  libEGL warning: failed to open /dev/dri/renderD128: Permission denied

  libEGL warning: NEEDS EXTENSION: falling back to kms_swrast
[... 236 lines cut ...]
          "reward_shaping": false,
          "camera_names": [
              "robot0_agentview_left",
              "robot0_agentview_right",
              "robot0_eye_in_hand"
          ],
          "camera_heights": 128,
          "camera_widths": 128,
          "has_renderer": false,
          "has_offscreen_renderer": true,
          "ignore_done": true,
          "use_object_obs": true,
          "use_camera_obs": true,
          "camera_depths": false
      }
  }

  Process 1 finished
```

### Evironment:
- OS: Windows WSL Ubuntu-24.04
- Python: 3.10
- numpy: 1.23.2
- Robominic: robomimic -b robocasa

snasiriany, 2024-10-05:
Do you know how many demos are you in `demotest.hdf5`? One possible suspicion is that the number of demos might be small in this case, leading to some unexpected issues. I wonder if you run the script in step 4 with `--num_procs 1` whether the issue goes away.

KysonYoshi, 2024-10-08:
Thank you for the solution. My demo.hdf5 file contains only one demonstration. After running the script from step 4 with `--num_procs 1`, the script was able to complete execution and returned control back to the command line. However, a new issue has arisen: when I try to access the im128.hdf5 dataset, it appears to be empty.

```
import h5py
import json

f = h5py.File(INSERT_DATASET_PATH)
demo = f["data"]["demo_1"]                        
obs = demo["obs"]                                 
left_img = obs["robot0_agentview_left_image"][:]  
ep_meta = json.loads(demo.attrs["ep_meta"])       
lang = ep_meta["lang"]                            
f.close()
```

```
Traceback (most recent call last):
  File "/home/cl6933/robocasa/show_replay.py", line 5, in <module>
    demo = f["data"]["demo_1"]                        # access demo 5
  File "h5py/_objects.pyx", line 54, in h5py._objects.with_phil.wrapper
  File "h5py/_objects.pyx", line 55, in h5py._objects.with_phil.wrapper
  File "/home/cl6933/miniconda3/envs/robocasa/lib/python3.10/site-packages/h5py/_hl/group.py", line 357, in __getitem__
    oid = h5o.open(self.id, self._e(name), lapl=self._lapl)
  File "h5py/_objects.pyx", line 54, in h5py._objects.with_phil.wrapper
  File "h5py/_objects.pyx", line 55, in h5py._objects.with_phil.wrapper
  File "h5py/h5o.pyx", line 241, in h5py.h5o.open
KeyError: "Unable to synchronously open object (object 'demo_1' doesn't exist)"
```

```
import h5py
import json

f = h5py.File(INSERT_DATASET_PATH)
demo = f["data"]                     
print(demo)                           
f.close()
```

```
<HDF5 group "/data" (0 members)>
```

snasiriany, 2024-10-17:
I think this is due to some issues with the multi core version of the dataset_states_to_obs.py script. We have an older version of this that using a single core: https://github.com/ARISE-Initiative/robomimic/blob/robocasa/robomimic/scripts/deprecated/single_core_dataset_states_to_obs.py

It's slightly deprecated now, but if you try this, does it at least successfully extract that single demo that you have?

kylekam, 2025-09-16:
> Thank you for the solution. My demo.hdf5 file contains only one demonstration. After running the script from step 4 with `--num_procs 1`, the script was able to complete execution and returned [...]

I am currently experiencing the same issue.