robocasa/robocasa #85: About generating and creating a new environment (layout file)

Abc12128, 2024-10-03:
Hi, thanks for designing such a great simulation environment. I want to create a new environment during my study. I want to generate a new layout.yaml file by importing some of my obj asset files and then doing the layout. But I don't quite know how to do it. How do you generate the layout.yaml file?

snasiriany, 2024-10-05:
We manually specify each layout yaml file! The existing ones are here: https://github.com/robocasa/robocasa/tree/main/robocasa/models/assets/scenes/kitchen_layouts

To create a new layout, I would recommend starting a new file in that directory, and then registering it here: https://github.com/robocasa/robocasa/blob/0d34893fdaf62d2b2a112908bafa37880b4c5011/robocasa/models/scenes/scene_registry.py#L7-L21

And then you can inspect / play with the layout by running the demo script `python -m robocasa.demos.demo_kitchen_scenes` and making adjustments to the yaml file accordingly.

Hope this is helpful!

Abc12128, 2024-10-05:
OK, thank you very much for your patient reply. Now I have some ideas about generating new environments. If I want to create a new environment, such as a bedroom, a living room, etc., I need to expand the asset library to meet the needs of the new environment. Can you give me some good suggestions so that I can go more smoothly?

Abc12128, 2024-10-09:
Thank you very much for opening this issue for me.
After your team closed the issue, I didn't see the button to reopen the issue for a while.

snasiriany, 2024-10-10:
There are many ways to go about this. One way is to take existing room scale scenes from other datasets / simulators and then convert them to the RoboCasa formats. I briefly shared some thoughts about this here: https://github.com/robocasa/robocasa/issues/9. It would be fairly involved and require expertise.

Another idea is to organically create these new types of scenes and populate them with objects. We did this already for kitchens, where we modeled 3D assets with articulation (sinks, microwaves, etc) and also procedurally generated ones (cabinets), and defined interfaces for them (see [here](https://github.com/robocasa/robocasa/tree/main/robocasa/models/fixtures)). This also requires putting these assets together in a new scene, and we do this programmatically for kitchens by defining [layout and style yaml files](https://github.com/robocasa/robocasa/tree/main/robocasa/models/assets/scenes), and also the [logic](https://github.com/robocasa/robocasa/tree/main/robocasa/models/scenes) that reads and assembles scenes based on these yaml files.

We laid the framework for how to do these things in kitchens and we want to extend this to other types of scenes. For the time being, the kitchen is very challenging so we want to explore the learning challenges there first before expanding to other kinds of scenes.

We do value open source contributions for those who want to help us expand the scenes though!

Closing for now, feel free to ask followup questions (I do read them).

closed 2024-10-10 by snasiriany

Abc12128, 2024-10-15:
Ok, thank you for your help.
Some time ago, I have been trying to use other simulation platforms to build a new scene layout (this step greatly reduces the time and difficulty of manual calculation), but the generated layout file is in json format. I tried to convert it to yaml format.
Before importing robocasa, I through  [this](https://github.com/kevinzakka/obj2mjcf) to generate xml files, but it seems to be different from the original object asset content in robocasa. 
I found that here #68 ，you mentioned that the original code has been modified, could you provide me with some help, so that I can generate the xml file that conforms to robocasa. This way I can better expand the scenes.

Abc12128, 2024-10-15:
Thank you again for your support of my work. Any help you can provide me would be greatly appreciated！

snasiriany, 2024-10-16:
> Could you provide me with some help, so that I can generate the xml file that conforms to robocasa. This way I can better expand the scenes.

Can you shoot me an email? My email is included on my CV on my homepage. I can share code privately. The code I can share with you regards the xml files for the objects, it's somewhat orthogonal to the scenes.

Abc12128, 2024-10-18:
> Can you shoot me an email? My email is included on my CV on my homepage. I can share code privately. The code I can share with you regards the xml files for the objects, it's somewhat orthogonal to the scenes.
Thank you very much for your willingness to help me. I think it will have a great positive impact on me.
I have contacted the email address mentioned in your resume soroush@cs.utexas.edu by email. I want to confirm with you whether you have received my message.

hahahalove57, 2024-10-20:
> > Can you shoot me an email? My email is included on my CV on my homepage. I can share code privately. The code I can share with you regards the xml files for the objects, it's somewhat orthogonal to the scenes.
> 
> Thank you very much for your willingness to help me. I think it will have a great positive impact on me. I have contacted the email address mentioned in your resume [soroush@cs.utexas.edu](mailto:soroush@cs.utexas.edu) by email. I want to confirm with you whether you have received my message.

Hey!I have been worked for translating obj file to XML file by modifying the code of obj2mjcf.if u need u can contact with me!^^

rivergold, 2025-03-10:
@hahahalove57 I want to know how to change obj2mjcf code, can you help me?