robocasa/robocasa #89: Support for motion planning?

gadorneles, 2024-10-09:
I would like to create my dataset using motion planning libraries with the Franka Emika Panda robot. Is there any example of something like that being used in robocasa?

snasiriany, 2024-10-10:
We don't offer motion planning out of the box with RoboCasa, but do offer the underlying OSC / IK based controllers (courtesy RoboSuite) that can be used in to execute plans generated by a motion planner.

There has been some interest in integrating motion planning with MuJoCo (eg. [here](https://github.com/google-deepmind/mujoco/discussions/2094)). There are some tools online for this but I am not sure which one best.

If you do implement a motion planner that works for RoboCasa/RoboSuite/or even MuJoCo, please consider sharing it with us! This would be very useful to the community.

closed 2024-10-10 by snasiriany

shaoyifei96, 2025-02-11:
@gadorneles Is there an update on this? We are recently looking for the same thing, gonna integrate curobo or maybe moveit

shaoyifei96, 2025-08-10:
No, we did get the geometry and did some local reactive obstacle avoidance, but we are not doing any global planning taking into account of all obstacles