robocasa/robocasa #91: Method to keep oobjects consistent each time the environment is initialized

KaiyuHu2001, 2024-10-09:
Some objects in the environment have their types and positions randomly generated. Is there a way to keep them consistent each time the environment is initialized?

snasiriany, 2024-10-10:
By design, we want to include randomization, but we do allow seed'ing environments, so that the environment resets are deterministic everytime you create and use an environment. See here: https://github.com/robocasa/robocasa/blob/0d34893fdaf62d2b2a112908bafa37880b4c5011/robocasa/environments/kitchen/kitchen.py#L219

In addition, we offer the ability to play back existing demonstrations with exactly the same type of objects and randomized positions that were used in the original demos: https://robocasa.ai/docs/datasets/using_datasets.html#inspecting-and-visualizing-datasets

snasiriany, 2024-10-10:
One thing to add: you can specify the exact path to a specific object that you want to sample, instead of a broad category, see here: https://github.com/robocasa/robocasa/blob/0d34893fdaf62d2b2a112908bafa37880b4c5011/robocasa/models/objects/kitchen_object_utils.py#L174

closed 2024-10-10 by snasiriany