robocasa/robocasa #98: Resolution of camera observation

ziyi105, 2024-10-27:
I am trying to save robot0_frontview as png images for further processing, the images saved have a square dimension, which is different from what i see on the onscreen render (rectangle). Can I clarify whether it is expected for the images to be saved as a square image or there is any setting that allows me to save what i see on the onscreen simulation. I am also wondering if there is anyway where i can make the robot invisible in the image saved. 

Here is my code to capture image:
```
    env = robosuite.make(
        **config,
        has_renderer=True,
        has_offscreen_renderer=True,
        render_camera="robot0_frontview",
        ignore_done=True,
        use_camera_obs=True,
        control_freq=20,
        camera_depths=True,
        camera_names="robot0_frontview",
        renderer="mjviewer",
        render_collision_mesh=False,
        render_visual_mesh=False,
    )
```

```
        if capture_images:
            image_dir = "robocasa\captured_views"
            os.makedirs(image_dir, exist_ok=True)
            image = obs["robot0_frontview_image"]      
            image_bgr = cv2.cvtColor(image, cv2.COLOR_RGB2BGR)      
            image_path = os.path.join(image_dir, f"step_{int(start):04d}.png")
            cv2.imwrite(image_path, cv2.flip(image_bgr, 0))
            print("Image capture complete.")
```

abhihjoshi, 2024-10-27:
I believe you can set the camera height and width when creating the env. Specifically, you can pass them as arguments: https://github.com/robocasa/robocasa/blob/370f986aa3934be6c134ecb978952423df9a1ed0/robocasa/environments/kitchen/kitchen.py#L213-L214
I believe these set the heights and widths of the camera observations. 

Regarding a camera from the top of the scene, you can define your own camera(s) in your arena xml as such: https://github.com/ARISE-Initiative/robosuite/blob/74981fd347660205efb4f8c0005e28ec621f5304/robosuite/models/assets/arenas/empty_arena.xml#L25

ziyi105, 2024-10-29:
Thanks for your insight!

closed 2024-10-29 by ziyi105