robocasa/robocasa #101: ImportError: cannot import name 'load_controller_config' from 'robosuite'

ModiShi, 2024-11-02:
seems that robosuite has updated the function load_controller_config after [a new commit](https://github.com/ARISE-Initiative/robosuite/commit/915dce5a528d256c52fa83c98e056c795ae093f2#diff-43d556d216248e717089d8c6c24441372f8501f45587c22714360ad572669bd2L1), which will results a ImportError  in file [env_utils.py](https://github.com/robocasa/robocasa/blob/main/robocasa/utils/env_utils.py)

snasiriany, 2024-11-03:
Fixed with this commit: https://github.com/robocasa/robocasa/commit/41d25b2d5c349044c82f324d61dba39c73bbafbe
We forgot to update env_utils.py to accommodate the new changes in robosuite.
Thanks for reporting this to us!

closed 2024-11-03 by snasiriany

qqz03, 2025-05-19:
Thanks for the update! Quick check — should **eval_utils.py** also be updated to use `from robosuite.controllers import load_composite_controller_config`?