robocasa/robocasa #102: ERRORS running python -m robocasa.demos.demo_tasks，KeyError: 'body_parts'

loubron23, 2024-11-12:
THANKS for the great work! When I run 'python -m robocasa.demos.demo_tasks', an error like this will appear. I am wondering if this problem is related to the mujoco version. When I installed the master version of robosuite, mujoco required version 3.2.5, but when I installed robocasa, it had to be version 3.1.1. Why is this version error? I installed it according to the tutorial. Are these packages updated?

(robocasa) loubron@loubron-PC:~/disk/robocasa$ python -m robocasa.demos.demo_tasks
[robosuite WARNING] Could not import robosuite_models. Some robots may not be available. If you want to use these robots, please install robosuite_models from source (https://github.com/ARISE-Initiative/robosuite_models) or through pip install. (__init__.py:30)
WARNING: mimicgen environments not imported since mimicgen is not installed!
Tasks:
[0] PnPCounterToCab: pick and place from counter to cabinet
[1] PnPCounterToSink: pick and place from counter to sink
[2] PnPMicrowaveToCounter: pick and place from microwave to counter
[3] PnPStoveToCounter: pick and place from stove to counter
[4] OpenSingleDoor: open cabinet or microwave door
[5] CloseDrawer: close drawer
[6] TurnOnMicrowave: turn on microwave
[7] TurnOnSinkFaucet: turn on sink faucet
[8] TurnOnStove: turn on stove
[9] ArrangeVegetables: arrange vegetables on a cutting board
[10] MicrowaveThawing: place frozen food in microwave for thawing
[11] RestockPantry: restock cans in pantry
[12] PreSoakPan: prepare pan for washing
[13] PrepareCoffee: make coffee

Choose an option 0 to 13, or any other key for default (PnPCounterToCab): 3
Initializing environment for PnPStoveToCounter...
[robosuite INFO] Loading controller configuration from: /home/loubron/disk/robosuite/robosuite/controllers/config/robots/default_pandaomron.json (composite_controller_factory.py:121)
Traceback (most recent call last):
  File "/home/loubron/anaconda3/envs/robocasa/lib/python3.10/runpy.py", line 196, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "/home/loubron/anaconda3/envs/robocasa/lib/python3.10/runpy.py", line 86, in _run_code
    exec(code, run_globals)
  File "/home/loubron/disk/robocasa/robocasa/demos/demo_tasks.py", line 148, in <module>
    playback_dataset(parser)
  File "/home/loubron/disk/robocasa/robocasa/scripts/playback_dataset.py", line 335, in playback_dataset
    env = robosuite.make(**env_kwargs)
  File "/home/loubron/disk/robosuite/robosuite/environments/base.py", line 41, in make
    return REGISTERED_ENVS[env_name](*args, **kwargs)
  File "/home/loubron/disk/robocasa/robocasa/environments/kitchen/single_stage/kitchen_pnp.py", line 825, in __init__
    super().__init__(obj_groups=obj_groups, *args, **kwargs)
  File "/home/loubron/disk/robocasa/robocasa/environments/kitchen/single_stage/kitchen_pnp.py", line 18, in __init__
    super().__init__(*args, **kwargs)
  File "/home/loubron/disk/robocasa/robocasa/environments/kitchen/kitchen.py", line 297, in __init__
    controller_configs = refactor_composite_controller_config(
  File "/home/loubron/disk/robocasa/robocasa/utils/config_utils.py", line 60, in refactor_composite_controller_config
    new_controller_config["body_parts"][arm] = new_arm_config
KeyError: 'body_parts'

YinpeiDai, 2024-11-12:
Same issues here...

snasiriany, 2024-11-14:
Sorry about that! A recent update to robosuite broke things. Looking to find a fix. Will be resolved today.

snasiriany, 2024-11-14:
Update: fixed! Please pull the latest code from:
* `main` branch of robocasa
* `master` branch of robosuite

closed 2024-11-14 by snasiriany