robocasa/robocasa #104: question about evaluation

LeiHanyue, 2024-11-21:
Thank you for your excellent work. I have a question about the evaluation.
During the experimental stage of the paper, it was mentioned that "For each task, we evaluate the model performance across 50 trials across five fixed evaluation scenes, each with a distinct floor plan and style," and the results for this evaluation are presented in Figure 7. Do the "five fixed evaluation scenes" here refer to the five val data segments within the hdf5 dataset collected by humans for each task? I noticed that there are also five scenes specified in the eval_utils.py file in the robocasa/robocasa/utils directory, which are ((1, 1), (2, 2), (4, 4), (6, 9), (7, 10)). I would like to know what these five scenes in the eval_utils.py file correspond to, or whether the "five fixed evaluation scenes" correspond to these five scenes. If so, are the positions of the objects and the robot randomized during the evaluation?

snasiriany, 2024-12-05:
There are a total of 120 possible scenes: a Cartesian product of 10 scenes and 12 styles, ie. 10x12 = 120.

> I noticed that there are also five scenes specified in the eval_utils.py file in the robocasa/robocasa/utils directory, which are ((1, 1), (2, 2), (4, 4), (6, 9), (7, 10))
We evaluate on the five scenes mentioned above, ((1, 1), (2, 2), (4, 4), (6, 9), (7, 10)). For example (6, 9) means layout 6 and style 9, and that constitutes one scene.

Note that during data collection and MimicGen data generation we withheld 2 styles (see how style 9 and 10 are missing here: https://github.com/robocasa/robocasa/blob/1e9321975ac16f12b88460dbc7096f8f4ca84135/robocasa/scripts/collect_demos.py#L387). Thus anything with style id 9 and 10 are considered unseen scenes for evaluation.

The positions of the object and robot are randomized during all evaluations.

closed 2024-12-05 by snasiriany

LeiHanyue, 2024-12-12:
In robocasa, can we get the same image observations when replaying recorded data, which probably includes making the camera and material textures the same as when recording the data. I found that the replay through the reset_to function currently supports restoring the simulation of the physical scene, which includes placing the same objects in the same position, setting the same layout and style, and placing the robot in the position when recording;

snasiriany, 2025-01-02:
yes, this is possible by default. are you encountering an issue where the textures / cameras are not the same when replaying?