robocasa/robocasa #107: Changing demo style

xuanshao27, 2024-12-04:


xuanshao27, 2024-12-04:
Can we change the demo's style when the demo is already collected? I want to train my policy in different styles, and I am already collected a hdf5 file with 300 demos, but all of their style is 8, can i change some of those demos' style into 1 or 2?

snasiriany, 2024-12-05:
Unfortunately, it's not possible to do this completely, since changing the styles would entail geometric changes that mean the recorded motions are no longer producing the same interactions.

However, you can consider randomly swapping out the textures of the walls/counter/cabinet/floor, as this is only a visual change. To do so, run the `dataset_states_to_obs.py` [script outlined here](https://robocasa.ai/docs/use_cases/creating_datasets.html#extract-image-observations), with the `--generative_textures` flag. This will give some additional visual diversity to your dataset.

closed 2024-12-05 by snasiriany