robocasa/robocasa #108: How can I generated data with mimicgen?

xuanshao27, 2024-12-10:
Thank your for your patient answer for my last question,I wander is there any tool that you write for generating human data with mimicgen?  I want to generate my own data by using mimicgen but I don't know how to do.

snasiriany, 2024-12-12:
There's now a fork of the official mimicgen repo for this: https://github.com/robocasa/mimicgen

It’s a three step process:
1. Extract metadata for the source human dataset, eg:
```
python mimicgen/mimicgen/scripts/prepare_src_dataset.py --dataset ~/robocasa/datasets/v0.1/single_stage/kitchen_doors/CloseDoubleDoor/2024-04-29/demo.hdf5
```

2. Extract MimicGen demos:
```
python mimicgen/mimicgen/scripts/generate_dataset.py --config mimicgen/mimicgen/exps/templates/robocasa/single_stage/kitchen_doors/CloseDoubleDoor.json
```

4. Extract image observations on the resulting dataset
```
OMP_NUM_THREADS=1 MPI_NUM_THREADS=1 MKL_NUM_THREADS=1 OPENBLAS_NUM_THREADS=1 python robocasa/robocasa/scripts/dataset_states_to_obs.py --dataset <ds-path>
```

Let us know if you have any questions!

closed 2024-12-12 by snasiriany

snasiriany, 2024-12-17:
Update: moved code to official MimicGen repo under new branch of `experimental/mimicgen`: `https://github.com/NVlabs/mimicgen/tree/experimental/robocasa`