robocasa/robocasa #111: Textures not loading correctly when trying to test policy on training demos enviornment

SruthiSudhakar, 2024-12-13:
Hi I see that the loading of the textures is happening in [edit_model_xml](https://github.com/robocasa/robocasa/blob/27f9924662dee3b534916e4288eee50189ca2c42/robocasa/environments/kitchen/kitchen.py#L1020) function in robocasa/environments/kitchen/kitchen.py. Even if I set self.generative_textures=False, in [this](https://github.com/robocasa/robocasa/blob/27f9924662dee3b534916e4288eee50189ca2c42/robocasa/environments/kitchen/kitchen.py#L1143) line of code (so that it does not edit the xml with randomly generated textures), the textures still do not match the observations in the demonstrations. This makes me wonder if the xml in the demonstrations (found by data['data']['demo_1'].attrs['model_file'] for example) have the right texture files. Though I might be missing something here. Please let me know!

snasiriany, 2024-12-14:
Hi @SruthiSudhakar thanks for raising this issue. One thing that's happening right now is that the identity of the generative textures is not being saved in the hdf5 demos correctly, eg. I see the following behavior for an hdf5 file with generative textures saved:

```
import h5py
import json

f = h5py.File("path")
json.loads(f["data/demo_1"].attrs["ep_meta"])["gen_textures"] # returns empty dictionary {}
```

We would rely on this information saved in the metadata to be able to load back and play back the demos with the same textures as what appeared originally in the demo. If this is a feature you'd like (consistent playback of demos with the same textures), then I can look into it and propose a fix early next week. Let me know.

SruthiSudhakar, 2024-12-15:
Ah ok. Yes this would be a very useful feature so that I can ensure policies are well trained and differentiate between poorly trained policies vs out of distribution data. A fix early this week would be extremely helpful, thank you so much!

snasiriany, 2024-12-17:
Just pushed a fix: https://github.com/robocasa/robocasa/commit/4eae2b22c53f6e55eb09105c9c6b7f8ab997d845

Please update to the latest code in main.

From now on, whenever you run image extraction [(see steps here)](https://robocasa.ai/docs/use_cases/creating_datasets.html#extract-image-observations), the meta data for the textures will be saved, allowing you to then reply the dataset with consistent textures.

Let me know if you have any followup questions!

closed 2024-12-17 by snasiriany

SruthiSudhakar, 2024-12-17:
Thank you very much, this is very helpful!