robocasa/robocasa #113: How to add other third-person cameras

BoyuLi-XH, 2024-12-18:
Thanks for the issue[https://github.com/robocasa/robocasa/issues/18](url). I run 
```
python ./robocasa/scripts/collect_demos.py --debug --env PrepareCoffee --robots GR1FixedLowerBody
```
successfully. But after I changed the robots to H1, the robot third person camera input is inappropriate, which is shown below.
[image]
This is the camera named "robot0_agentview_center". Other cameras such as 'robot0_agentview_left', 'robot0_agentview_right', 'robot0_frontview', 'robot0_eye_in_hand' cannot observe the whole robot state. How can I add a suitable third-person camera?

BoyuLi-XH, 2024-12-18:
Another question: Is there any control policy or language skill that I can control the humanoid robot to move or interact with the objects?

BoyuLi-XH, 2024-12-23:
Hope to get your reply @snasiriany

snasiriany, 2024-12-24:
Hi @BoyuLi-XH this is because the camera settings may need to be adjusted for each different robot. Go into this file: https://github.com/robocasa/robocasa/blob/45eb87eaca84eaac1d3ac0185338fc10bbe1554f/robocasa/utils/camera_utils.py#L94-L95

And add a section for the H1 robot with any camera adjustments that suit your case.

> Another question: Is there any control policy or language skill that I can control the humanoid robot to move or interact with the objects?
We provide datasets and policy learning code, which you can use to learn some language conditioned policies. Please see our documentation page for more information: https://robocasa.ai/docs/introduction/overview.html

closed 2024-12-24 by snasiriany